arendst/Tasmota #1488: Instead of counting uptime of a POW, count the working seconds of a connected device

lobocobra, 2018-01-02:
This is not an issue report. I will document my path to solve following problem of me. Which might be of use for other users.

**What I intend to do:**
The heating of my house is running with oil. I want to use SONOFF POW to measure how much Oil is left in the tank and how much oil I use a day/week. 
=> wWhen POW detects higher watt than 200w, it shall start to count the run-time of the heating in seconds and shall save it on the POW (like the uptime). Like this I can calculate how much oil the heating used and how much is left in the tank.

In case you have already realized this or have an easy idea to implement it, then please post your remarks here. I will start to change the uptime counter in the code in the way. Once it works, I will paste the changed code parts as I did it for my shutter project (which works btw very nicely).

lobocobra, 2018-01-02:
Hmm I am somehow stuck in the code....
=> If I want to check in SONOFF, if Power is higher than the limit, do I need to launch
      boolean EnergyCommand(char *type, uint16_t index, char *dataBuf, uint16_t data_len, int16_t payload)

Did anyone something similar? I want to count the seconds the PowerHigh was active and save this value and to send it from time to time via MQTT.
=> But I am stuck as I have no experience, as the energy code is part of a library. So I would need to launch a function from the main program... but which?

PLEASE SOMEONE HELP

davidelang, 2018-01-02:
power is measured every loginterval seconds, and each time it's measured, it's 
sent out via mqtt.

the sort analysis and tracking over time that you are talking about are really 
best done in your home automation software.

lobocobra, 2018-01-02:
Hmm thanks for the hint, but I will continue to search how I get access on that variable in SONOFF tab. 
=> once I have that, I can count up the seconds its running at multiply it with the l/h of oil burned.
=> I will avoid issues when internet is down, like this

Just need to find out, how tho get the value inside the sonoff tab... grrrrr

lobocobra, 2018-01-02:
Ok, found a way to count it in the library itself. Now I have no clue how to use the Webserver. Why there is always a "D_" in front of variables to display? I would normally assume them to be variables to be replaced by translated text, but I see no where any code, that does this job?

Just want to display on the webpage the seconds.... anyone a clue?

lobocobra, 2018-01-03:
Now I have all pieces together except:
1) how can I send out a MQTT message?
    void MqttPublishDirect(const char* topic, boolean retained)
    => How do I use it.... just add a topic as a string?    
2) how can I add on the web server an entry on the front page, what is the command to display?
     func += F("}1" D_BOOT_COUNT "}2"); func += String(Brenner_Uptime);
    => why is there a D_xxxxxx everywhere? 
3) how do I save a value I generate in the code, so that it is kept?
    snprintf_P(log_data, sizeof(log_data), PSTR(D_LOG_APPLICATION D_BOOT_COUNT " %d"), Settings.bootcount);
    => this line seems to save data, but how I create my own, and how I read it on start?

**Please help**

Frogmore42, 2018-01-03:
Two sayings come to mind:
1. Pushing on a rope
2. Working uphill
Since you seem to want to use MQTT, you should just do the calculation in the thing that is getting the MQTT message.  In node-red it would be pretty easy to do the calculation/accumulation.  Any changes you make to the code will need to be done again to any new versions of tasmota that you need/want.

lobocobra, 2018-01-03:
Any centralized solution will depend on many components (MQTT/WIFI) to work, and will end up in wrong calculations on how much time the heating is on, if any component is temp. offline.
=> In addition, once working there is no need to update the firmware anyhow and if yes, it's easy to repeat.

So do you know how to use the functions? If yes your support is welcome. 
=> If now, I will find it out, document here and will help somebody else with his uphill solution as done already previousely.

lobocobra, 2018-01-03:
I solved point 1):
`MqttPublishDirect(const char* topic, boolean retained)`
Point 2) I found a work around with:
`func += F("}1" D_BOOT_COUNT "}2"); func += String(Brenner_Uptime);`
=> I will use an old sonoff basic to find out how to change the variable text
Point 3) I could use boot - count until I found the final solution.

I will omitt to find out how the "D_" story work for the moment and concentrate on my other 2 points.

lobocobra, 2018-01-03:
Found the background about the D_ variables. As thought, they are simply the stand in for the translated text. Go to the Arduino project into the language folder. In there you see 1 file per language. If you need additional Variables you can add them there!

davidelang, 2018-01-03:
on the other hand, if you write every value to flash, you will wear out your 
flash fairly quickly, and if you don't, then a restart of the device will loose 
some history.

perfect is the enemy of good enough.

lobocobra, 2018-01-05:
I will save the data locally only at 12h and send out the current status, when requested by MQTT or at least at midnight.
=> like this I can loose maximal 12h of data

Other existing solutions, struggle with the problem that if they miss a "heating stop" message (power down, etc), the whole data is corrupt. 
=> my solution might miss some data, but will never have wrong/corrupt data 
Anyhow I will calculate only how much oil is left in the tank, so that will be good enough.

STILL SOME HELP NEEDED
While I can send out by MQTT the seconds the heating was on, I still struggle to get out a full message, when requested. Can anyone help me to code the MQTT message, which can be requested by the MQTT server?

davidelang, 2018-01-05:
On Fri, 5 Jan 2018, lobocobra wrote:

> While I can send out by MQTT the seconds the heating was on, I still struggle 
> to get out a full message, when requested. Can anyone help me to code the MQTT 
> message, which can be requested by the MQTT server?

you would have to implement a new command, this would be in the sonoff.ino file.

you can look at how the existing commands are implemented.

goatchurchprime, 2018-01-06:
I've been posting a similar request to this here:
   https://bitbucket.org/xoseperez/espurna/issues/369/report-energy-use-per-event-on-the-emon

Basically, we don't want a series of point measurements separated by a number of seconds.  Instead we want the device to fit the measurements to a square wave profile and report the square's dimensions, because these will relate to the events that we are interested in.

davidelang, 2018-01-06:
Well, the Pow hardware can only do point measurements, and each measurement 
takes several seconds to make, so to start with you only have point measurements 
every X seconds

the place to do the more complex math you are looking for is in your home 
automation tool, not in the sonoff.

Frogmore42, 2018-01-06:
As has been pointed out before, it is best to let Tasmota do what it does best and let the higher levels do what they can do better.  Set up the Tasmota device to report power use every 10s, use Node-red or something else to aggregate the time that the power is above whatever threshold you want.  Now you have a solution that is using Tasmota in the way it is designed to be used and still getting the information you want.  But, there is always more than one way to achieve any goal, so do it the way that makes you happy 😀

lobocobra, 2018-01-06:
Basically I have almost all working. 
=> I can set MaxPower and count the seconds while POWER is higher than x seconds
=> I can save data locally once a day
=> I am partially able to send MQTT messages, but struggle to send out one on request.

Does anyone know how MQTT can be extened by 1 or 2 variables? I spend far to much time currently for something that one you you might know right away.

Once I have all done I will post it here. 
@goatchurchprime, I agree with you, that would be perfect, but let's see, maybe we can do this.

lobocobra, 2018-01-07:
Ok here is what I did, it works for me perfectly and other might profit from my work... 
=> I can set over mqtt a Watt level, from which the seconds above this Watt is counted
=> I an read the current value on the web page
=> I can request the value by MQTT
=> I know how to reset the reboot counter without resetting all data (and thus how to save data)
      (not implemented below)
First line is always the unchanged code, in order to have a reference

**SONOFF.ino line 126**
```
int uptime = 0;                             // Current uptime in hours
//lobocobra start
long Brenner_Uptime =0L;
// lobocobra end
```
line 2663
```
  Settings.bootcount++;
  // lobocobra working reset  bootcount
  // Settings.bootcount =1 ;
```
**User_config_h** line 37
```
//#define MY_LANGUAGE            it-IT           // Italian in Italy
// lobocobra start
#define D_BRENNER "Brenner Sekunden"
//lobocobra end
```
**Webserver line 1487**
```
  func += F("}1" D_BOOT_COUNT "}2"); func += String(Settings.bootcount);
//lobocobra start
  func += F("}1"  D_BRENNER "}2"); func += String(Brenner_Uptime);
  String RRS1 = "Brenner Sekunden:";
  String RRS2 = RRS1 + Brenner_Uptime;
  MqttPublishDirect(RRS2.c_str(),false);
// lobocobra end // Show in MQTT the value, when you go on the website
```
**xsns_03_energy.ino line 28**
```
#define FEATURE_POWER_LIMIT  true
//lobo start
#define D_JSON_Brenner "BrennerSekunden" 
//lobo end
```
**line 154**
```
      energy_daily = (float)energy_kWhtoday / 100000000;
     //lobocobra: increase Device uptime every second the used power is higher than powerhigh
      if (energy_power > Settings.energy_max_power) { Brenner_Uptime++;  } //powerhigh is used to set the trigger
     // end lobocode
```
**line 891**
```
  char energy_yesterday_chr[10];
//lobo start
  char energy_lobo_sekunden_chr[10];
//lobo end
```
**line 914**
```
  dtostrfd((float)Settings.energy_kWhyesterday / 100000000, Settings.flag2.energy_resolution, energy_yesterday_chr);
  //lobo start
  dtostrfd(Brenner_Uptime, Settings.flag2.voltage_resolution, energy_lobo_sekunden_chr);

  if (json) {
    snprintf_P(speriod, sizeof(speriod), PSTR(",\"" D_JSON_PERIOD "\":%s"), energy_period_chr);
    snprintf_P(mqtt_data, sizeof(mqtt_data), PSTR("%s,\"" D_RSLT_ENERGY "\":{\"" D_JSON_TOTAL "\":%s,\"" D_JSON_YESTERDAY "\":%s,\"" D_JSON_TODAY "\":%s%s,\"" D_JSON_POWERUSAGE "\":%s,\"" D_JSON_POWERFACTOR "\":%s,\"" D_JSON_VOLTAGE "\":%s,\"" D_JSON_CURRENT "\":%s,\"" D_JSON_Brenner "\":%s}"),
      mqtt_data, energy_total_chr, energy_yesterday_chr, energy_daily_chr, (show_energy_period) ? speriod : "", energy_power_chr, energy_power_factor_chr, energy_voltage_chr, energy_current_chr,energy_lobo_sekunden_chr);
//      {"Time":"2018-01-07T01:06:24","ENERGY":{"Total":15.533,"Yesterday":2.972,"Today":0.110,"Period":0,"Power":290,"Factor":0.84,"Voltage":228,"Current":1.506,"BrennerSekunden":19}}
  //lobo end    
#ifdef USE_DOMOTICZ
```

lobocobra, 2018-01-07:
I will update and close the topic as soon I have it bit nicer and all included and working in openhab.

lobocobra, 2018-03-30:
It's working nicely by now, and I will soon update my progress. I have ONE remaining issue, for which I do seek an advice/help....

I want to save some of the variables 3x per day. With 100k minimal write cycles possible and knowing that SYSCFG does a rotate, I should not run into problems to wear the HW. But here comes my question....
=> I want to extend SYSCFG by some UNSIGNED LONG variables. Can I just add them at the end? I see that SettingsSaveAll does check the size of SYSCFG, but I see also that behind each variable arendst did write the HEX numbers. WHY that? Was it just to check how much mem is used, or did I miss in the code something? 
**==> Can I add 8 unsigned long variables just at the end?**

arendst, 2018-03-31:
To the latest 5.12.0i version you can add your longs as you expect. The hex numbers are there to keep score of the real memory addresses used by the processor; having a byte (8 bits) next to a uint16_t (16 bits) will lead to an extra byte in between if the byte is located on an even address as the compiler/processor only allows 16-bit memory access. A long (32 bits) will always need to be at an address divisable by 4bytes.

That's why you see the free gaps in the settings area.

As I just added the 16 timer longs at the end you are safe to add your longs next to them.

lobocobra, 2018-03-31:
Ahhh yes of course! 

Many thanks arendst for your detailed explanation. Btw... your SW is impressive and your coding style even more!
=> I will port my changes from 5.1 to the latest version and then publish here my solution, before I close the topic. I hope it helps some others.