arendst/Tasmota #1489: Different switch behavior

battistaar, 2018-01-02:
Is it possible to configure a switch to directly control the relay and another to send a MQTT message?
I'd like to connect GPIO14 to the wall switch and GPIO3 to a PIR sensor but I don't want the PIR sensor to directly control the relay.

davidelang, 2018-01-02:
no, switchtopic applies to all switches

alfredopironti, 2018-01-09:
Are there any plans to add a SwitchTopic<x> command, much like SwitchMode<x> works, so that each switch can have its own topic and work differently?

I have a different use case that would benefit from such improvement. For the record, I'm trying to drive a motor pump. This will have a physical 3-way selector: Manual OFF; Auto (homeassistant timer); Manual ON. I could disconnect wires on Manual OFF, and bypass the sonoff in Manual ON, but in this case I'd loose tracking of the real state in homeassistant.

Hence, I'd like to define two switches on the sonoff, the first that follows Manual OFF state, and publishes on a Manual_OFF topic; the second that does the same on Manual_ON. In this way homeassistant can reliably track state.

Or maybe there's a simpler way to achieve the above that I'm missing? Thanks for the support and the hard work!

davidelang, 2018-01-09:
you can already define the multiple switches to react differently

but they would all send the same topic, your home automation software can 
translate what is sent by the switches into the actions that you want to take.

It's unlikely that there will be multiple switch topics, it's not needed if you 
use a home automation software (which is what Tasmota is designed for), and it 
eats up a noticable amout of memory to support (which e are short of)

alfredopironti, 2018-01-09:
Thanks for the superfast answer!

How would I distinguish which switch was activated from the home automation software? Would each switch append a \<x> to the topic, so that in fact I can tell which switch was activated?

davidelang, 2018-01-10:
switch1 will send POWER1
switch2 will send POWER2
etc

alfredopironti, 2018-01-10:
Ah fantastic!

This perfectly covers my use case, and it seems to makes sense for the general case, too.

Re: the origina post, @battistaar if you're using some home automation software to handle the PIR switch, maybe you can configure it to turn on the light when the wall switch sends its MQTT message?

battistaar, 2018-01-11:
If my WiFi goes down or I’m changing configuration on my PI or HA i’m not
able to turn on and off the light.
The wall switch should directly control the relay and work in almost every
situation. The PIR should send MQTT to HA so I can customize the
automations.
I’ll probably go with custom firmware, it’s sad because Tasmota is great,
but I can’t depend on the software, not everyone in the house is used with
this technology. If the light won’t turn on they’ll probably change the
bulb (or call me)

davidelang, 2018-01-11:
if you don't set switchtopic, and the switch number is the same as the relay 
number, it will control the relay without needing the mqtt server

note that there is a bug that while the device is trying to connect to a mqtt 
server, nothing else can happen, including responding to things locally (this is 
an arduino library bug, not a tasmota bug)

mkh595, 2018-01-16:
_Is it possible to configure a switch to directly control the relay and another to send a MQTT message?_

It is possible with combination of button and switch. See #1274

mkh595, 2018-01-16:
_If my WiFi goes down or I’m changing configuration on my PI or HA i’m not
able to turn on and off the light.
The wall switch should directly control the relay and work in almost every
situation. The PIR should send MQTT to HA so I can customize the
automations._

I think this is not correct behavior, see #1614

Mobyn, 2018-04-03:
Hi There, I'm not able to get a differentiation between POWER1 and POWER2. Im Running 5.12.0 on NodeMCU with 09 Switch1 and 10 Switch2 configured. I can change the switch modes but can't seem to get a different MQTT state message for each switch. 

02:51:41 MQT: tele/d1tass/STATE = {"Time":"2018-04-03T02:51:41"...}}
02:51:41 MQT: tele/d1tass/SENSOR = {"Time":"2018-04-03T02:51:41","Switch1":"ON","Switch2":"ON"}
02:53:31 MQT: stat/d1tass/RESULT = {"POWER":"ON"}
02:53:31 MQT: stat/d1tass/POWER = ON
02:53:31 MQT: stat/d1tass/RESULT = {"POWER":"OFF"}
02:53:31 MQT: stat/d1tass/POWER = OFF
02:53:32 MQT: stat/d1tass/RESULT = {"POWER":"ON"}
02:53:32 MQT: stat/d1tass/POWER = ON
02:53:32 MQT: stat/d1tass/RESULT = {"POWER":"OFF"}
02:53:32 MQT: stat/d1tass/POWER = OFF

closed 2018-07-10 by ascillato2