arendst/Tasmota #1491: other ESP8266 hardwares with Tasmota

jcoudrais, 2018-01-02:
Hi,

Shloud we get other hardwares compatibility with your tasmota scripts ?

Some cheap smart plugs seem to have same hardware and it will be great to get automation without cloud for that...(Noduino, and variants Tuya, Jinvoo, Voion, Xenon)

Best regards

davidelang, 2018-01-03:
Tasmota will run on just about anything that's ESP8266 based. If it's a common 
module, there may be a module definition for it, but in most cases you just 
select a wemos module and define the GPIOs as appropriate for how your device is 
wired.

jcoudrais, 2018-01-04:
Hi,

Thank you for your answer, I tried that but I can't get the hidden ssid of this tuya smart plug... Can't find any useful too...
Any idea ?

webtop, 2018-01-04:
I have a number of raw ESP8266 12 modules flashed with ST and have no problems whatsoever with the firmware (5.10). I have the WeMo D1 selected as David mentions. The ESP on a Sonoff is massively underutilized.

davidelang, 2018-01-04:
opening it up and connecting directly to the esp will always work.

anything else is going to involve reverse engineering the update method your 
vendor decided to implement. (as a practical matter, most of them don't care 
about security and implement a version of an open-source tool, but it still 
takes work to figure which one and what modificatins they have made to it)

jcoudrais, 2018-01-05:
Hi, thanks for your answers,

As the plug is in factory reset, I can see an hidden wifi that is opened, but I don't know it's ssid, so I can't connect to it....

I tried several tools like reSSIDer but no one can give me the ssid name.

Rgds

jcoudrais, 2018-01-05:
Well, that was not the hidden SSID the problem... there is a ssid named smartlife-CE0A (4 last bytes from MACAD)... but the process to upgrade is not compatible it seems... 
:-(

webtop, 2018-01-06:
If it has an ESP8266, it's compatible. You're not exactly upgrading, but
completely overwriting the current firmware. If you can get access to the
ESP chip, you're in business

hwarang098tkd, 2018-01-23:
if I flash tasmota software to a homemade module with esp 8266 who to determine the gpios pin numbers. Because I have to use something with different pins numbers. like esp -01

davidelang, 2018-01-24:
you determine what pin number is what (by how you wire things to the ESP)

Use the wemos module type as it allows you to assign any function to any GPIO 
pin.

closed 2018-06-08 by stale[bot]