arendst/Tasmota #1492: Flicking Light

merlin051, 2018-01-02:
Hi,

I'm using pin14 and switch mode 9 to work with a rocker switch which is fine, but depending on other noise on the circuit the light flickers on/off regardless of the current state.

Is there any option that can introduce a 250ms delay to verify its actually changed state instead of the false readings from the circuit noise?

My log looks like this:

22:31:19 MQT: stat/sonoff1/RESULT = {"POWER":"ON"}
22:31:19 MQT: stat/sonoff1/POWER = ON
22:31:19 MQT: stat/sonoff1/RESULT = {"POWER":"OFF"}
22:31:19 MQT: stat/sonoff1/POWER = OFF
22:31:21 MQT: stat/sonoff1/RESULT = {"POWER":"ON"}
22:31:21 MQT: stat/sonoff1/POWER = ON
22:31:21 MQT: stat/sonoff1/RESULT = {"POWER":"OFF"}
22:31:21 MQT: stat/sonoff1/POWER = OFF
22:31:31 MQT: stat/sonoff1/RESULT = {"POWER":"ON"}
22:31:31 MQT: stat/sonoff1/POWER = ON
22:31:31 MQT: stat/sonoff1/RESULT = {"POWER":"OFF"}
22:31:31 MQT: stat/sonoff1/POWER = OFF
22:31:59 MQT: stat/sonoff1/RESULT = {"POWER":"ON"}
22:31:59 MQT: stat/sonoff1/POWER = ON
22:31:59 MQT: stat/sonoff1/RESULT = {"POWER":"OFF"}
22:31:59 MQT: stat/sonoff1/POWER = OFF

merlin051, 2018-01-02:
Forgot to mention this is a sonoff basic using the latest firmware.

My switch is wired to gnd and gpio14,  GPIO Sensor is configured to 09 Switch1.

I've read about adding a resister on the gnd pin but I'm not sure where to start with that.

merlin051, 2018-01-29:
anyone?

lobradov, 2018-01-29:
This has been already addressed number of times - for example, check #850.

//_ on phone

closed 2018-06-13 by stale[bot]