arendst/Tasmota #1494: Blind controller remote emulation

nosaj66au, 2018-01-03:
Hi All, 
   Have been playing with a motorised blind / curtain controls from China. They appear to be branded under various names world wide but google "Dooya" you'll get the idea. Anyway I've hack the protocol which is a 40bit RF 433mhz data stream and have it working on a Uno and compiling on the ESP8266. Next step is integration to Tasmota I now use Tasmota exclusively and don't want to build an orphan. 
Q1, How many people are interested blind control?
Q2, Where would I put it? Is it a remote or in the RF Bridge or new device? It could also be a user pin device as it just requires a RF Transmitter and could be mounted on a Sonoff device really easy.

Jason

nosaj66au, 2018-01-03:
Update: Just reviewed the code and I may create an RFSend device/cnmd just like IRSend containing a JSON packet. It makes sense considering the number of options Remote ID (24bits), Blind(8bits), Action(8bits)  comments please?
 Could also implement RFrecv would be good for weather stations and rain level transmitters.

closed 2018-05-17 by stale[bot]

jk2020, 2018-09-02:
Would be very interested to get more information how a Dooya DC90 remote control can be connected to tosmota flashed sonoff rf bridge

SantiagoSotoC, 2018-10-20:
I'm in the same situation as jk2020