arendst/Tasmota #1499: NOTICE: Tasmota is ready for release 2.4.0 of esp8266/Arduino

arendst, 2018-01-03:
Tasmota is based on esp8266/Arduino.

A few days ago **esp8266/Arduino version 2.4.0** has been released. Tasmota and it's supporting libraries as available in the lib folder will compile and run just fine.

Among many fixes this version also uses more flash space which makes it _almost_ impossible to easily OTA update Tasmota. A solution was developed months ago and involves an intermediate step using the sonoff-minimal.bin image available in all latest releases.

Now that this step has _almost_ become mandatory it opens up the possibility to add some more functionality to Tasmota in the future.

pascalsaul, 2018-01-03:
I just did a big bang upgrade to 5.10.0h with 2.4.0 from 5.10.0e with 2.3.0 and without issues. This includes all kind off Sonoff models :)

ChrSch84, 2018-01-03:
"add some more functionality to Tasmota in the future."

Is there any Plan for new funktions? And what could you add?

davidelang, 2018-01-04:
On Wed, 3 Jan 2018, Theo Arends wrote:

> Tasmota is based on esp8266/Arduino.

Arduino IDE, not Arduino hardware (just to clarify)

tobox, 2018-01-04:
I recompiled 5.10.0h with platformIO and esp8266v2.4.0 and everything seemed to be running properly. I was able to switch the relay properly and MQTT worked as well. 
But I was not able to access the web page (browser displays blank page and waits infinitely). 

Using MQTT I was able to start an update and I went back to my build from yesterday and everything is fine again. Any ideas?

Maybe related to this report?
https://github.com/arendst/Sonoff-Tasmota/issues/1347

pascalsaul, 2018-01-04:
I can access the webpages fine with Chrome and toggle the relays. With IE I can't toggle the relay very well.

tobox, 2018-01-04:
Strange. It does not work with Firefox neither with chrome on linux; but it works with a command line browser (i.e., w3m).

closed 2018-05-18 by stale[bot]