arendst/Tasmota #1500: Setting TEMP_CONVERSION in user_config.h does not produce Farenheit Temperature output

Nargousias, 2018-01-04:
Setting TEMP_CONVERSION in user_config.h does not produce Fahrenheit Temperature output.  

Setting I am using:
#define TEMP_CONVERSION        1                 // [SetOption8] Return temperature in (0 = Celsius or 1 = Fahrenheit)

Output I am getting on console:
{"Time":"2018-01-03T21:35:56","DHT11":{"Temperature":24.0,"Humidity":36.0},"TempUnit":"C"}

Have tried with a DHT11 and a DS18B20 with the same results.

ionciubotaru, 2018-01-04:
On the web interface press CONFIGURATION than RESET CONFIGURATION
After configuration reset you will see Fahrenheit Temperature output

profesionalTI, 2018-01-04:
You no need to modify the source code, anything Sonoff config or setup must be direct from console command. If you need to setup from Cº to Fº, enter the following command on console (web): 

cmnd/{mqtt topic name default is sonoff}/SetOption8 1

In my case is:

cmnd/swe0ff/SetOption8 1

With this command on web console, you will config the sonoff. I make many custom config to my sonoff and work very fine.

To more sonoff command visit: https://github.com/arendst/Sonoff-Tasmota/wiki/Commands <https://github.com/arendst/Sonoff-Tasmota/wiki/Commands> great help.

Regards

Nargousias, 2018-01-04:
Thanks,  That did the trick.  Working OK now.  Also thanks for the command link.

closed 2018-01-04 by Nargousias

profesionalTI, 2018-01-04:
Great. 😃

Coordialmente,
Best Regards,

Marcelo A. Anuch Medina
Consultor de Empresas Público/Privadas
Ingeniero Comercial, MBA UDD
Business Management

🔒   La información contenida en esta transmisión es confidencial y está dirigida exclusivamente a el o los destinatarios indicados. Cualquier uso por otras personas que no sean sus destinatarios está sancionado por la ley chilena. Si ha recibido esta transmisión por error, por favor destrúyala y notifique al remitente si corresponde. Las opiniones expresadas en este correo son las de su autor y pueden no ser compartidas por la compañía. Marcelo A. Anuch Medina,  no asume obligaciones ni responsabilidades por el contenido de este correo. Tampoco garantiza que la transmisión de este correo sea segura o libre de errores, y en consecuencia, no responde por virus, pérdidas, destrucción, recepción tardía o incompleta.

🍀  Antes de imprimir este correo electrónico asegúrese que sea realmente necesario: Cuidemos el medio ambiente! Todos vivimos en un solo planeta.