arendst/Tasmota #1501: Flashing Issue (solved) --> hints: (1) reboot, (2) try --no-stub in esptool.py

hs06, 2018-01-04:
Dear all,
just opening a ticket to document this solved issue so it may help others. It did cost me some time to solve it.

If you build with platform.io and get messages like this, the following may help you.

    Uploading .pioenvs/sonoff-minimal/firmware.bin
    Uploading 434704 bytes from .pioenvs/sonoff-minimal/firmware.bin to flash at 0x00000000
    warning: espcomm_send_command: didn't receive command response
    warning: espcomm_send_command(FLASH_DOWNLOAD_DATA) failed

    warning: espcomm_send_command: wrong direction/command: 0x01 0x03, expected 0x01 0x04`

I have tried a number of things, suspected the FTDI USB device (sanity check: connect RX and TX and open a terminal, e.g. ` screen /dev/tty.usbserial-A9EXTF3J 115200` and see if you get an echo).
However, despite the fact that I received the echo, I was not able to get any response from the ESP in boot/flash-mode. I downloaded esptool.py and experimented with no result. The connect always timed out:

```
$ esptool.py -p /dev/cu.usbserial-A9EXTF3J -b 9600 erase_flash
esptool.py v2.2
Connecting........_____....._____....._____....._____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header

```
However, what has helped me was indeed **a REBOOT of my system** (MacOS X 10.13.2).

However, I was not there yet. After the reboot, the esptool.py recognized the ESP, but failed after "Uploading stub...".

```
$ esptool.py -p /dev/cu.usbserial-A9EXTF3J -b 9600 erase_flash
esptool.py v2.2
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Uploading stub...

A fatal error occurred: Timed out waiting for packet header

```

Also the platform.io toolchain failed with the same result as stated above.

What has helped me to finally flash the image **was the option --no-stub of the esptool.py**. Both erasing and flashing worked perfectly with this option.

```
    $ esptool.py --no-stub --baud 9600 --port /dev/cu.usbserial-A9EXTF3J write_flash -fs 1MB -fm dout 0x0 ./Tasmota_Git/Sonoff-Tasmota/.pioenvs/sonoff-minimal/firmware.bin
    esptool.py v2.2
    Connecting....
    Detecting chip type... ESP8266
    Chip is ESP8266EX
    Enabling default SPI flash mode...
    Configuring flash size...
    Erasing flash...
    Took 1.65s to erase flash block
    Wrote 435200 bytes at 0x00000000 in 484.6 seconds (7.2 kbit/s)...

Leaving...
Hard resetting...
```

Of course, you can take a much higher baudrate - I was trying to be very cautious with my FTDI.

I hope this will help you if you experience similar issues.

Best
Hendrik

closed 2018-01-04 by hs06

thiagoarreguy, 2018-01-08:
Im with the same problem trying to connect to sonoff basic by FTDI to USB in my Mac OS X 10.13.2:

esptool.py --no-stub --baud 9600 --port /dev/cu.usbserial-A50285BI write_flash -fs 1MB -fm dout 0x0 sonoff.bin
esptool.py v2.2
Connecting........_____....._____....._____....._____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header

hs06, 2018-01-08:
Your device does not connect at all.
Try the following:
- check if 3v3 are supplied, not 5v
- check the FTDI by connecting rx with tx and see if you get an echo in a terminal 
- swap rx/tx
- reboot
- check that you correctly wired to the Sonoff
- check if you depressed the button

thiagoarreguy, 2018-01-08:
Its all right.
How can i get an echo in a terminal ?

thiagoarreguy, 2018-01-08:
A little evolution:
After push the button and while keep pushing, i unplugged USB, so i connect the to the USB port and stop to push the button.

esptool.py --no-stub --baud 115200 --port /dev/cu.usbserial-A50285BI read_flash 0x00000 0x100000 sonoff_backup_1.6.0.bin
esptool.py v2.2
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Enabling default SPI flash mode...

A fatal error occurred: ESP8266 ROM does not support function read_flash.

thiagoarreguy, 2018-01-08:
And it works for me:

esptool.py --no-stub --baud 115200 --port /dev/cu.usbserial-A50285BI write_flash -fs 1MB -fm dout 0x0 sonoff.bin
esptool.py v2.2
Connecting...
Detecting chip type... ESP8266
Chip is ESP8266EX
Enabling default SPI flash mode...
Configuring flash size...
Erasing flash...
Took 2.64s to erase flash block
Wrote 496640 bytes at 0x00000000 in 54.2 seconds (73.3 kbit/s)...

Leaving...
Hard resetting...