arendst/Tasmota #1504: Tasmota Firmware upgrade/update (steps prior to compile)

ScottyDS, 2018-01-04:
Hi Folks
This is not an issue, but an info share, on what was initially a problem for me.

My Sonoff/Tasmota landscape is growing significantly.   I am now up to 17 devices in 3 months. 
(Thanks again Theo and all on this forum for your marvelous work.)

This expanding landscape started causing me some "sub version" upgrade confusion and config pain with regards to the Sonoff folder in the Arduino IDE. 
I could not find any instruction on how to perform the end to end upgrade, with existing settings retention, so thought to document the steps I follow and invite comment /advice on the process. (perhaps I have been doing the upgrades incorrectly)

**The upgrade pain I experienced.**
 I upgraded from 5.8.0.c to 5.8.0.p, then again to 5.8.0.q.  Both times I used the upgrade instruction and openHAB's "bonus" Upgrade all Sonoffs using OTA. Both times I was left with all the Sonoff devices having been reset to default configs. It was quite messy having to go back and reconfigure the MQTT server, topics and friendly names, IP address config info etc.  (I use static IP's as my router can only supply 8 address reservations.) Also I made the other discipline mistake in not first testing on a single device before before deploying the  upgrade etc,etc.

**How I tried to improve my upgrade process .**
So what I do now is have a single portable IDE installation, but with a device specific folder in the sketches folder for each Sonoff device. (See image below) This allows me to create separate ino.bin files that I then copy to the web server for OTA updates and upgrade the devices. This approach does however amounts to 17 sketch folders that must be updated when the a new subversion is released.   So I keep a seperate, duplicate folder structure with only the customised user_config.h  files which I copy back once the folders have been updated with the new firmware.

The biggest risk is that the user_config.h may have also been updated in the new version.  I use a program called Diffuse to quickly check the differences between an existing default user_config.h and the new one. If there are differences, I then create 17 new user_config.h files.

**To summarise, My device upgrade process is.**
1. Download latest firmware and follow the instructions for installing and configuring a new portable IDE with Sonoff folder for an initial upload. 
2. Copy the 17 folders from the old IDE, to the new one, and copy the contents of the newly created potable/sketchbook/sonoff folder,  to all the device folders (attached image)
3.Copy the backed up/duplicate user_config.h folder structure to these the updated folders, thus overwriting the default user config.h files.
4. Run a little Dos script to change the sonoff.ino file names to match the sketchbook folder names.
5. Open the first sketch, compile it, and copy the resultant ino.bin file to the Raspi web server,
6. Open the second sketch, compile it and copy the resultant ino.bin file to the Raspi web server, rinse and repeat until all sketches have a new ino.bin file loaded on the Raspi web server.
7. Test the upgrade on a single Sonoff, and if it workks
8. Do a bulk upgrade on the remaining devices.

[image]

pascalsaul, 2018-01-04:
I'm using PlatformIO and my process is:

1. Download the complete .zip
2. Extract the zip
3. Rename user_config.h to user_config_org.h
4. Copy my customized user_config.h to the newly extracted folder
5. Compare user_config.h and user_config_org.h and merge changes manually
6. Build the new firmware
7. Copy the .bin to the webserver
8. Test new version on a individual device (mosquitto_pub -t 'cmnd/basic/Upgrade' -m '5.10.0h')
9. Upgrade all Sonoffs with the group topic (mosquitto_pub -t 'cmnd/sonoffs/Upgrade' -m '5.10.0h')
10. Watch update progress (mosquitto_sub -t '#' -v | grep Version)

ScottyDS, 2018-01-05:
Perfect, it seems I am on the right track then. Now how do we get these 20 processes included in the upgrade wiki? Will make it easier for new folks.

dweston, 2018-04-18:
Pardon my ignorance @pascalsaul but where do I find the '.bin' file referred to your point 8. above?