arendst/Tasmota #1505: Sonoff Basic and WS2812 - stuck on color schemes

UserXYZ, 2018-01-04:
Hello.
Happy New Year to everyone, thanks to @arendst and others for really cool and usable custom firmware for out Sonoff devices.

Since it is a time of festivities, I wanted to use my Sonoff Basic switch to add a light strip made of 30 WS2812 leds to have some fun with built-in color schemes and patterns, by configuring the module to have the strip on Rx pin.
They are quite nice and great fun, but they have a serious problem - they kill the CPU, thus making the switch totally unresponsive quite fast.
So, when I configure my Sonoff basic to publish on a certain mqtt topic, it works just fine until the moment I activate one of the color schemes. After that, within minutes (if not seconds), Sonoff stops reacting to any mqtt commands and stops publishing some of its status messages to designated mqtt topic.
What is interesting, it still manages to push those big status messages with all the data, time etc, but fail to push basic POWER(1/2) status messages.
In the end, web interface and mqtt messaging fail, the switch is totally uncontrollable, and I can't turn off the light pattern scheme by issuing "Power2 Off", so the only way to fix it is to kill the power to the switch,  and quickly after the start, issue "Power2 Off" through web console or mqtt.

Has anybody seen this behaviour? Am I doing something wrong, maybe I did't turn off something?

joecotton, 2018-01-04:
Are you powering the LED strip with the Sonoff's 3.3V supply?

UserXYZ, 2018-01-04:
No, with external power supply, as is supposed...

UserXYZ, 2018-01-10:
Any help guys? Has somebody tried this and maybe got some positive results?
It's not a product killer, but it would be nice to have it working properly if it's in the code...and looks most entertaining...

Sunrise17, 2018-02-04:
Is it possible to power ws2812 (till max 3number of pixels) with 3.3V from Sonoff basic onboard?

handsomejackuk, 2018-02-11:
I have two ws2812b leds connect to the 5v rail, i dont know how many it can support, but anyone brave enough to try more than 2 let me know, the location of this can be found in the schematic but its across d5 diode, the gnd can be  picked up in lots of places but the 5v only seems to be easily accessed at the back of a sonoff basic pcb, check my youtube video..

https://www.youtube.com/watch?v=CPw5ixEF89Y