arendst/Tasmota #1509: Sonoff RF Bridge and Domoticz Integration help

Joeyhza, 2018-01-04:
Hi,

Hoping someone can shed some light on how to integrate a Tasmota flashed RF Bridge in to Domoticz.
I have 433mhz door sensors and PIR sensors which the bridge learns about successfully.

I just have no idea how to get it all in to Domoticz - I'm sure this has been done before but for the life of me I can't seem to figure this out. I just want to know if a door is open or closed and if and when movement is detected etc.

Any help appreciated.

Thanks

zarss, 2018-01-05:
I have same problem. Googling around all evening.

Joeyhza, 2018-01-06:
Same for me - been researching the hell out of this :)

Not having much luck - let me know if you get anywhere with it please

declan2511, 2018-01-13:
I have used node-red and translated the MQTT message by changing the topic to Domoticz/in & out.  It worked but isn't ideal.  Still exploring other methods.

Joeyhza, 2018-01-18:
Decided to use Espurna for my bridge for now. It had IDX assignments for all 16 switches - easier Domoticz integration.

Will reflash with Tasmota for consistency if/when IDX is implemented for RFBridge on Tasmota.

declan2511, 2018-01-18:
Good idea, I will give it a go! Thanks

closed 2018-06-01 by stale[bot]