arendst/Tasmota #1510: Support Rain Sensor YL-38 FC-37

FabianHM, 2018-01-04:
I am interested in the rain sensor YL-38 FC-37. Would it be possible to support this sensor in one of the next versions?

davidelang, 2018-01-04:
On Thu, 4 Jan 2018, FabianHM wrote:

> I am interested in the rain sensor YL-38 FC-37. Would it be possible to support this sensor in one of the next versions?

You are far more likely to get the device supported if you can provide 
documentation on what the interface to the device is rather than asking the 
developers to do the research to find it (or better still, provide code that 
talks to the device)

FabianHM, 2018-01-05:
Here is more information from another forum about this component:

[http://www.instructables.com/id/Arduino-Modules-Rain-Sensor/](url)
What information is still needed?
Would be glad if the implementation in tasmota would be possible....

arendst, 2018-01-05:
Looks like analog data but max 5V. With some resistors you can connect it to a wemos and use the current tasmota software to output the analog value. Monitor margins in your home automation tool.

closed 2018-05-19 by stale[bot]