arendst/Tasmota #1511: Amazon Echo potential issue - turning on all devices when looking at 'smart home' section

warst, 2018-01-04:
Hi,

I am rather new to Tasmota (about 24 hours new!) but have managed to flash 2x Basic's and 1x Touch and have them all working with HASSIO and Alexa. My issue is that whenever I am on the Alexa web portal and click on 'Smart Home' then 'Devices', my three Sonoff modules are turned on. All are using 5.10.0 

This wouldn't be so bad however one module is connected to an extension lead which has the heated blanket attached. I don't want it accidentally turning on and being left on all day!! 

Does anyone have any ideas?

CultUr, 2018-01-05:
I have the same issue.

warst, 2018-01-05:
In fact, my electric blanket module turned itself on last night.... Woke me up in a sweat, it was lovely - not!!

merlin051, 2018-01-05:
I've recently just done the same and u don't have this issue.

I'm using Mqtt on HA to control the sonoff and get the power state. To get the device to show in Alexa I'm using hue_emulation on HA. (https://home-assistant.io/components/emulated_hue/)

The emulation on the sonoff is set to None.

Give this a try?

reloxx13, 2018-01-05:
its fixed in the dev build.

warst, 2018-01-05:
So if I flash the dev build I won’t wake up in a hot sweat when I go to sleep tonight?! Or do I have two issues here as only the Blanket turned on, not the lamp and they are set up identical.... Does the dev build fix the Alexa switching everything on issue only?

Kind Regards,
Warren Tack.
Technical Solutions Manager
Network 2 Supplies LTD
Mob: 07775687998

reloxx13, 2018-01-05:
correction: alexa auto turn on devices by visiting the app smart home devices list fixed in dev build.

merlin051, 2018-01-05:
I'm running Home Assistant 0.60.0 and I don't see this issue, but my config is slightly different

warst, 2018-01-05:
@reloxx13 I couldn't figure out how to flash a dev build, ota (without compiling it myself and I'm a bit pushed for time) so @merlin051 as per your suggestion, I removed the emulation from the sonoff switches and added the hue_emulation: into home assistant config file and they are showing & working with Alexa now, lets see how it goes!

warst, 2018-01-05:
I have also since built the latest development firmware and uploaded it to the switches, just as a learning curve. Used Atom on OSX, surprisingly simple, I like it! This helped a lot - [https://github.com/arendst/Sonoff-Tasmota/wiki/Beginner-Guide---Create-your-own-Firmware-Build](url)

closed 2018-05-19 by stale[bot]