arendst/Tasmota #1513: Sonoff T1 UK  RF Pairing

juan11perez, 2018-01-05:
NOT AN ISSUE, but wanted to share to save others some time. This is the way to pair the T1 with an RF remote. http://ewelink.coolkit.cc/?p=698.

Problem is that once you've loaded Tasmota, pressing the button long, resets tasmotta and I was unable to link with the switch tirelessly again. Tried the default SSID (indebuurt1) with no luck; so I had to pull the switch and reflash it.

So the advise is simple. Pair the RF before flashing tasmota. Once flashed it will ratain the RF code.

arendst, 2018-01-05:
Or use ``SetOption13 1`` to disable long press

juan11perez, 2018-01-05:
thank you. where do i do that? and how would I pair the RF?

arendst, 2018-01-05:
Pairing will be the same as with the original software as it is not serviced by tasmota but by the "other" microcontroller.

All commands can be given by either MQTT, Serial or web console. See wiki for more commands.

juan11perez, 2018-01-05:
Thank you very much. so for my benefit. I can SetOption13 1 throught the console, but I can still do the RF pairing afterwards without resetting tasmotta?

arendst, 2018-01-05:
Correct

juan11perez, 2018-01-05:
Once again, thank you. and thank you for sharing this fantastic software.

closed 2018-01-05 by arendst