arendst/Tasmota #1514: Retrieving sensor data via HTTP request

Anx2k, 2018-01-05:
Is there some mechanism by which you can get the current sensor data (ideally in JSON format) from the current device?  I've looked through the webserver code, but nothing jumped out at me - and I know it supports MQTT, but I really just want to setup something to periodically grab the data - and ideally not have to scrape it from the /ay page.

davidelang, 2018-01-05:
no, the primary use case for Tasmota is MQTT, the webserver is intended for 
initial configuration and troubleshooting, everything there is intended for 
human consumption.

you may be able to abuse the console interface, but I think you will have to do 
parsing of the output of that as well.

webtop, 2018-01-05:
I'm confused by the use-case. If you can use HTTP, why not use MQTT? What are your restrictions? Where are you wanting to call the device from?

If you present more conditions we might be able to help better. For example, you can use node-red to act as a relay and convert the MQTT output to JSON, but not sure if you can support node-red.

Anx2k, 2018-01-05:
My point wasn't to alter the primary use case at all - it was just the point that by slightly modifying how the webpage polls to display data could easily open up a whole different set of options, with no real cost and greater compliance with standards (as opposed to how /ay currently outputs the data.

In terms of use case, I have a little over 150 Sonoff Pow's that I'm using to monitor different devices - and I only need to spot-check the Pow's as a secondary data source.  There's a ton of traffic rolling around on the MQTT side, that I was hoping to eliminate since I only need to check devices periodically, not have live detailed state.  While I suspect there aren't that many people using as many devices I have, I have to believe there's others who would love to have direct JSON data without introducing another point of failure into their system - or for people who might only want to have one or two devices, and don't want a whole bunch of extra things running just to facilitate a pretty trivial request.

arendst, 2018-01-05:
As documented in the wiki you have to ability to execute any command from http. As of the latest devs it will output valid JSON only. So I see no problem at all.

Example to retrieve pow status 10 information with ``http://pow/cm?cmnd=status%2010`` returns
```
{"StatusSNS":{"Time":"2018-01-05T18:26:09","ENERGY":{"Total":25.746,"Yesterday":0.185,"Today":0.148,"Power":8.0,"Factor":0.41,"Voltage":209.7,"Current":0.094}}}```

Anx2k, 2018-01-06:
That last command looks like EXACTLY what I'm after - I tried it on mine though, and it only returns this:

STATUS10 = {"StatusSNS":{"Time":"2018-01-06T05:56:24"}}

You mentioned the latest dev, so I'm assuming this isn't on 5.10.0, but instead on the latest checkins?

Anx2k, 2018-01-06:
Just a followup - I tries this running the same software on the TH16, and I got this:

STATUS10 = {"StatusSNS":{"Time":"2018-01-06T06:13:05","AM2301":{"Temperature":79.9,"Humidity":17.2},"TempUnit":"F"}}

Both show the correct data in the WebUI - and idea why my POW would return no extra data, but the TH16 would?

arendst, 2018-01-06:
As said before, major changes in the latest dev. Just install and enjoy.

Anx2k, 2018-01-06:
Awesome - thanks for your help and the great project!

closed 2018-01-06 by Anx2k