arendst/Tasmota #1516: Support Dust Sensor PMS5003

isvedic, 2018-01-05:
Hi,

It's becoming more and more useful to have some dust data for automating ventilation.
Would it be possible to support this sensor in one of the next versions? 

I know Sonoff SC have GP2Y10 but that only outputs voltage and can't get info on dust particles size

All info and some code examples can be found at: https://learn.adafruit.com/pm25-air-quality-sensor

I'm already using sensor SHT3x for ventilation logic when to power it and with this it would be more practical for workshops and living areas

Thanks for great work!

webtop, 2018-01-05:
What's the raw output for that? You can probaly use the UART-Tx and UART-Rx pins to communicate with the device, but you still have to do something with that raw data.

isvedic, 2018-01-05:
9600 baud data streams out once per second from sensor 

Posting it to mqtt on topics for PM1.0, PM2.5 and PM10.0 concentration

arendst, 2018-01-05:
Must be possible. Send me one and I will look into it ;-)

isvedic, 2018-01-06:
Your shipping address?

arendst, 2018-01-06:
Send me a pm via the github start page.

arendst, 2018-01-06:
Send me a pm via the repository home page.

closed 2018-02-25 by isvedic

PEPZ, 2018-04-04:
Hi, @isvedic , did you manage to arrange testing of the PMS5003? There is currently the PMS5003ST module, which adds temperature and humidity to the particle measurement. Would be great to have this supported as well.

catfish34, 2018-11-22:
Do somebody know how to connect  PMS5003 to sonoff?
As I understand PMS5003  need 5V power. But on sonoff has 3.3?

isvedic, 2019-01-15:
Yes, you need only DATA pin connected to sonoff, be sure you are using right pin, configure sonoff and restart

catfish34, 2019-01-29:
I found in Internet this screen:
http://cs5-2.4pda.to/12384783.png

I tried to configure my sonoff basic but nothing changed on main page when I changed configuration.
My video:
http://recordit.co/UdfOJluMEs

Can it be problem with my PMS5003  sensor?

catfish34, 2019-01-29:
It is very strange... because if I configured this device to PZEM 004 or another sensor I can see values with 0.
My video:
http://recordit.co/kCeXxsws1S

Possible PMS5003 is special sensor. I will try to buy new one.

@isvedic Thank for your help.

isvedic, 2019-01-29:
Try another pin on sonoff and be sure you use correct DATA pin on PMS sensor

dan-cristian, 2019-03-11:
Works for me, I used Generic and Serial In set as PMS5003(69).
I compiled the firmware with #define USE_PMS5003.