arendst/Tasmota #1519: Platformio stuck in Serial Monitor

wjh84, 2018-01-05:
Hi all,

I'm pretty new to all this and still figuring out how everything works.
I bought myself 2 Sonoff basic switches, one with date 2017/5/15, and one with date 2017/6/20.

I used atom with platformio to flash the Tasmota software to them. Flashing works fine, but then I try open the Serial Monitor and get no result.

Also tried to Flash with Arduino IDE, same result, flashing worked fine, but nothing in the serial monitor.

Tried also to erase the flash with ESPtools, and then flash again with platformio, but no result.

Attached a screenshot of where platformio got stuck.

Hope anyone can help!

Jan
[image]

webtop, 2018-01-05:
I'd say you're not getting any telemetry data, not the PIO is "stuck". Can you log into the device from a browser?

wjh84, 2018-01-06:
No, I don't now the IP address. It should come from the serial monitor right?

webtop, 2018-01-07:
If you left the AP IP at 0.0.0.0, then it will use DHCP and request an IP from your router. You can then check your router to see what IP is assigned to your device. Then reserve that IP in your router.

PlumbumDN, 2018-01-07:
Please Type "SerislLog 4" in the Console and hit return. So the log-level is raised to 'debug more'

Ans dose tue Console echo something when you press tue button in the Sonoff?

PlumbumDN, 2018-01-07:
I ment 'SerialLog 4'

wjh84, 2018-01-08:
I used the 0.0.0.0 IP address so IP was made automatically. He's definitely not connected to my router.

Also my COM port is not available if I not put it in Flashing Mode (holding the button when plug the USB in)

I can't really find where to type commands in a console here... Sorry I'm very new to this.

Could it be that my USB to TTL device isn't working properly?

Thanks a lot so far guys!

Jan

PlumbumDN, 2018-01-08:
But flashing succeeded, right?

You can type a command, clicking in the PowerShell area (black part), the type it (you don't see anything) and hit "return". This should generate some output, if the sonoff and ttl-usb are working....

wjh84, 2018-01-08:
I get this when I type in the PowerShell area:
---------------------------------------------
PS D:\SmartHome\Sonoff-Tasmota-development> SerialLog 4
SerialLog : The term 'SerialLog' is not recognized as the name of a cmdlet, function, script file, or operable program. Check
the spelling of the name, or if a path was included, verify that the path is correct and try again.
At line:1 char:1
+ SerialLog 4
+ ~~~~~~~~~
    + CategoryInfo          : ObjectNotFound: (SerialLog:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException

PS D:\SmartHome\Sonoff-Tasmota-development>
---------------------------------------------

Guess I do something wrong..?

PlumbumDN, 2018-01-08:
You need to be connected to the ESP like in your screenshot. Then you can send the command to the sonoff/ESP

wjh84, 2018-01-08:
Ok thanks.
Tried that, but nothing happens, I begin to think more and more my USB to TTL is broken.
Also because the serial monitor only sees the COM port when my ESP is in flashing mode.. Or is that normal?

webtop, 2018-01-08:
If it sees your ESP in flash mode, your TTL cable is fine.

wjh84, 2018-01-11:
Sorry guys, this was a beginners mistake.

For referral if anyone might have a similar problem:
I didn't put my jumper on the 5V to VCC.

The Serial Monitor works fine now, thanks anyway!

closed 2018-05-25 by stale[bot]