arendst/Tasmota #1520: Suggestion/ Request: Tasmota firmware for Geekcreit 5V 1 and 4-channel

rankerson, 2018-01-05:
** I stated the same issue in the [SonOTA github](https://github.com/mirko/SonOTA/issues/92) **

Hello together,

this is a question/ request concerning the below mentioned devices from Greekcreit (see links below). Especially the _5V-DIY-Four-Channel_ seems to be similar or equal to the _Sonoff 5V-DIY-Four-Channel_ and both are using ewelink as Cloud solution.

I tried to flash them by soldering and OTA, but both failed.
**_Flashing by soldering:_**
I was not able to connect to the device (assuming that I soldered the correct pins on the device)

**_Flashing by OTA:_**
(with python, following the instructions on https://haus-automatisierung.com/hardware/sonoff/2018/01/04/sonoff-vorstellung-part-11.html)
Initial connect is possible, but after changing in origin wifi the device is not getting the image from my host.

Does anyone has a clue how to flash those devices or is it planned to develop a Tosmota image and procedure to flash such devices?

Thanks in advance!

Cheers
Sascha

**Geekcreit 5V-DIY-Four-Channel**
https://www.banggood.com/Geekcreit-USB-5V-Or-DC-7V-32V-DIY-4-Channel-Jog-Inching-Self-locking-WIFI-Wireless-Smart-Home-Switch-p-1157460.html?rmmds=detail-left-hotproducts__1&cur_warehouse=CN

**Geekcreit 5V-or-DC-7V-13V-DIY-Four-Channel**
https://www.banggood.com/USB-5V-Or-DC-7V-13V-DIY-One-Channel-Jog-Inching-Self-locking-WIFI-Wireless-Smart-Home-Switch-p-1065181.html?rmmds=detail-top-buytogether-auto__2&cur_warehouse=CN

**Sonoff 5V-DIY-Four-Channel**
https://www.banggood.com/5V-DIY-Four-Channel-Jog-Inching-Self-locking-WIFI-Wireless-Smart-Home-Switch-p-1065305.html?rmmds=search&cur_warehouse=CN

davidelang, 2018-01-05:
If it is running an ESP8266, then you can flash it with a wired connection.

What you need to do is make sure that it's in flash mode at boot. Normally this 
just consists of making sure that GPIO0 is connected to ground when the system 
boots.

I saw a post yesterday where someone found that there were a couple other pins 
that had to be set to a specific value to get into the boot mode.

beyond that, just be sure you have your serial adapter hooked up properly (and 
that it's a 3.3v device). This means tx on the adapter is connected to rx on the 
ESP. It's very common for people to connetc tx-tx and rx-rx and fail to flash.

rankerson, 2018-01-06:
Hi David,

thank you for your reply. I am not quite sure, if it is running on ESP8266, but think so.

I know, that we have to take care about grounding the GPIO0, but we do not have a clue, which one it is as they are not named on the board. Concerning tx and rx we now as well and we tried both connections always assuming that we found the correct pins ... I illustrated it in the attached picture:

We assume that the 4 pins within the square are relevant, because ...
... **Pins in A** are marked with Key1 to x and I think those are the pins for manual buttons to control the different switches.
... **Pins in B** are for the "433MHz receive board" to connect.

Within the square we measured the voltage and found out that it is number 1. Therefore we assumed number 4 for GPI0 (Ground) as the usual combination is sth. like voltage, rx/tx, tx/rx, ground on the other Sonoff boards.

Maybe someone can correct or confirm the used combination? Where did you found the page concerning the boot mode?

Cheers
Sascha

[image]

milkProvider, 2018-02-26:
I was able to flash this board by directly soldering wires to ESP8266 but it seems like something is still wrong. I tried two versions of tasmota, the one that works on my other sonoff boards and the latest versions. I also tried directly uploading Tasmota to the board using esptool but I was not able to start WIFI server, it seems like board was constantly resetting after 5-10s. After that I tried uploading firmware using arduino IDE with preconfigured WIFI settings but the board failed to connect to my router, not sure why. One thing that I see is that when Tasmota is already uploaded to the ESP8266, the diodes that are under relays blinks, this is not natural (as they would be wired to tx/rx as behaviour is similar) and the light is very dimmed.

glumbi, 2018-03-07:
Grüsse, ich habe es am Wochenende geschafft mit sonota zu flashen. erst dachte ich es funktioniert nicht, doch beim zweiten Versuch lief es durch. ich habe dann auf Sonoff 4CH Pro umgestellt und die Firmware auf 5.12 geupdatet. Funktioniert bisher problemlos.

rankerson, 2018-03-10:
Hi Glumbi,

would be helpfull to get some details, what you did in order to successfully flash the device.

Cheers
Sascha

taste66, 2018-03-10:
Got it working flashing Tasomota v5.12.0 using these instructions to get the serial working http://www.ablegen.de/itead4channel/
Selected Sonoff 4CH Module in configuration and I can control all relays.
It is not ESP8266 but ESP8285 based but as far as I can tell this looks compatible.