arendst/Tasmota #1522: #include <TasmotaSerial.h>

Jerby13, 2018-01-05:
I tried to compile Tasmota update v5.10.0h and v5.10.0i in Arduino IDE I get fatal error of missing TasmotaSerial library. 

> ...\Sonoff\Sonoff-Tasmota-development v5.10.0i\Sonoff-Tasmota-development\sonoff\xsns_03_energy.ino:252:27: fatal error: TasmotaSerial.h: No such file or directory
__
 #include <TasmotaSerial.h>

                           ^

compilation 

> terminated.

I'm very impressed about the time and effort you put in updating and expanding the Tasmota package. Thanks a lot.

arendst, 2018-01-05:
Copy all libraries from the repository lib folder and you should be fine.

Jerby13, 2018-01-06:
Do not know why I didn't check the library folder for the TasmotaSerialI, stupid me. Thank to your quick respons I have version 5.10.0i up and running now :-) 👍

closed 2018-01-06 by Jerby13

MRAnderson1221, 2018-01-13:
i am receiving the same error.  can you explain a little more what files i am grabbing and copying.
i have a folder called tasmotaSerial-1.0.0 in a lib folder.  i copied the tasmotaserial.h into the sonoff folder, but that didnt work.

thanks

dinger69, 2018-01-13:
I downloaded the latest from the lib repository and replaced the original's but it still will not compile with the same error:

C:\Sonoff-Tasmota-5.11.0\sonoff\xsns_03_energy.ino:252:27: fatal error: TasmotaSerial.h: No such file or directory

 #include <TasmotaSerial.h>

                           ^

compilation terminated.

exit status 1
Error compiling for board Generic ESP8266 Module.

Danie10, 2018-01-13:
I'm getting the identical error with Tasmota 5.11. I have copied the TasmotaSerial.h file into /Sonoff-Tasmota-5.11.1/sonoff/ but the error still comes up. @Jerby13 what did you copy to where exactly?

dinger69, 2018-01-13:
I copied the latest lib directory from the repository and overwrote my
existing /Sonoff-Tasmota-5.11.0/lib.

Full disclosure...I am a noob so I could easily be missing something
obvious.

arendst, 2018-01-13:
I suppose you have read the wiki regarding how to setup and use Arduino IDE with Tasmota: https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE

dinger69, 2018-01-13:
Thanks arendst. I was able to correct my problems by updating the ESP8266 Board version to 2.4.0. I also commented out several of the failing components that I don't need at the moment such as (CO2 sensors, Energy Sensors, etc.).

Danie10, 2018-01-13:
Thanks @arendst I think I must try that Wiki rather as I had the exact same problems with doing it in Windows. I did come right eventually by downgrading all versions to match exactly what was in DrZzs's video at https://www.youtube.com/watch?v=chyVjtYb0EA&index=5&list=PLL02pNmOcO9mADQt-naxoTPYW-mvwZwsW. But I'm still not managing to see the web server although my Sonoff Basic is connecting to my Wifi with an IP address. I think I'll follow your advise and reflash using that Wiki (I know more now than I did this morning so hopefully it goes better). But at least I'll have a different issue for a different thread.

Danie10, 2018-01-14:
@arendst quick feedback. The wiki worked like a charm (Linux users need to give permission to write to the USB port with "sudo chmod 666 /dev/ttyUSB0"). After I got a "host unreachable" error when pinging and the fixed IP never showed either, I realised my problems were probably network router related all along. Rebooted router and used plain DHCP and the switch is working just fine with webserver screen appearing. It's often the case we only look where we expect to see the fault....

mtabbut, 2018-01-18:
I'm having the same trouble

/Sonoff-Tasmota-5.11.1/sonoff/xdrv_03_energy.ino:252:56: fatal error: /lib/TasmotaSerial-1.0.1/src/TasmotaSerial.h: No such file or directory
 #include </lib/TasmotaSerial-1.0.1/src/TasmotaSerial.h>

mtabbut, 2018-01-18:
I'm having the same trouble

/Sonoff-Tasmota-5.11.1/sonoff/xdrv_03_energy.ino:252:56: fatal error: /lib/TasmotaSerial-1.0.1/src/TasmotaSerial.h: No such file or directory
 #include <TasmotaSerial.h>

I am running arduino on Mac OS 10.13.  I tried re-copying all the lib files off the repository.  I do have a TasmodaSerial.h in the /lib/TasmotaSerial-1.0.1/src/ folder.  Any other thoughts?  Thanks.

crozo5, 2018-02-04:
I'm having the same problem too...

/Sonoff-Tasmota-5.11.1/sonoff/xdrv_03_energy.ino:252:27: fatal error: TasmotaSerial.h: No such file or directory
 #include <TasmotaSerial.h>
                           ^
compilation terminated.
exit status 1
Error compilando para la tarjeta Generic ESP8266 Module.

And also running macOS

judders, 2018-02-10:
Not sure about Mac or PC. Linux here.
You need to copy all of the Sonoff-Tasmota -development/lib folders into the arduino/lib folder.

Jerby13, 2018-02-11:
If you follow judders advice you should have TasmotaSerial.cpp and TasmotaSerial.h files in directory 
"....\arduino-1.8.5\libraries\TasmotaSerial-1.1.0\src"
In that directory you should also see other folders of libraries.
Restart IDE after you have copied the files.

Cattierfriend, 2018-11-08:
I have the same error, if I open the sonoff.ino file in the tabs across the top I can clearly see the TasmotaSerial.h file is there but every time I run the verification same problem again.  I've read that wiki page backwards and forwards 10 times I just cant see what I'm missing.

judders, 2018-11-09:
@Cattierfriend Make sure you download the "master" release. Not the "development" release.