arendst/Tasmota #1523: Reading the analog pin A0 on Wemos D1 Mini

phdemartin, 2018-01-05:
Hi
I'm triing to get my Wemos D1 Mini to read the voltage value on the analog pin A0 of my Wemos D1 Mini.
Not an essy task to write a skech in the Arduino IDE with just a fue documentation available.
Would it be possible to implement it in Sonoff-Tasmota ?

davidelang, 2018-01-05:
It's already available in tasmota. It's a compile time option to use it to 
report the supply voltage or as an independent input.

phdemartin, 2018-01-05:
Ok thanks, I did not find it in the "user_config.h" ¿ Where can I enable it ?
So, I could use it to sense the value of an external voltage from a voltage divider for example:
My goal is to monitore a solar system, and I nead a way to log the voltage fluctuaction of the system.
If I've got it right from http://esp8266.github.io/Arduino/versions/2.0.0/doc/reference.html , the voltage sensing capacity is 0 to 1.0 volt ?

webtop, 2018-01-06:
If you want to do that, you'll need to comment out the line in the user_config.h file that reads something about USE_ADC. Comment that out and the A0 pin will act as an analog input for 0.0 to 1.0VDC

phdemartin, 2018-01-06:
It's working great thank's a lot, 
I've put a voltage divided to lower my 12 volt imput to a managable value for the esp8266 and I can now simply aply some math to my analog readingstring to get my value.
The strange trhing is that the A0 dont getfro 0 to 1024 at 0v to 1.0v, but get to 1024 at 3.3, contrary to the doc...

Frogmore42, 2018-01-06:
NodeMCU defines a voltage divider on the analog input to bring the range up to 3V3, so what you are seeing is expected.

phdemartin, 2018-01-06:
Exactly.
If it's useful fore someone.
I've used this voltage divider calculator to set my resistance value: http://www.ohmslawcalculator.com/voltage-divider-calculator
It gave me the 10kohms for R1 and 3.3kohms for R2 to reach a value lower of 3.3 volts for a 12 volts input voltage.
As I don't use MQTT in this system, I simply retrieve the 0 to 1024 result with curl, format the http output to isolate the needed string, apply a simple division to get my input voltage value. 
Here is the bash script to log the result.
> #! /bin/bash
while true ; do
read=$(curl -s http://192.168.0.35/cm?cmnd=status%2010 | cut -d":" -f6 | cut -d "}" -f1)
voltage=$(echo "scale=2; $read / 52" | bc)
time=$(date +"%F %R:%S")
clear
echo $voltage
echo "$time $voltage" > ~/voltage.log
sleep  60
done

mkh595, 2018-01-07:
_Not an essy task to write a skech in the Arduino IDE with just a fue documentation available._

Why is not easy? Arduino IDE provides example, also interent is full of them, it is just few lines of code

```
pinMode(A0, INPUT);
raw = analogRead(A0);
volt=raw/1023.0;
volt=volt*3.3;
```

Wemos D1 Mini has already build in divider R1 220k/ R2 100k for pin A0. That's why you can connect 3.3V to pin A0 even datasheet for ESP8266 says that max voltage for A0 is 1V. You only need add resitor in series with R1 to increase max voltage range. For 12V has to be R1 1100k. So 1100k minus 220k (already there) is 880k. You don't need to build another divider R1/R2.
[Here is explanation with example](https://arduinodiy.wordpress.com/2016/12/25/monitoring-lipo-battery-voltage-with-wemos-d1-minibattery-shield-and-thingspeak/)

phdemartin, 2018-01-07:
Ok.
I did not see any example in my IDE, my mistake, I've tried to work based on the analog arduino skech without success, and I would also need to setup a web-server interface to extract the datas... but anyway, as I also want to integrate some meteorological sensor in that installation, the Tasmota firmware offered me the complete package I needed and way more easy for me.
I did not know that there was already a voltage divider present in the D1 mini, I'll experiment with that.
Thank's

closed 2018-05-21 by stale[bot]