arendst/Tasmota #1525: Unable to access wifi network after changing wifi Channel

TommySharpNZ, 2018-01-06:
I changed my router so that the channel was not "Auto" and made it channel 8. All my devices reconnected except for all my tasmota flashed devices. I didn;t change the SSID or password...

I even changed the channel back to "auto" and I just cannot get them to reconnect.

Any help would be appreciated.

TommySharpNZ, 2018-01-06:
Okay after about 10 or 15 save and reboots they all seem to have reconnected again.

Makes me think it would be good to be able to have a password on the AP that gets spun up when it can't connect.... Someone could see the AP connect to it, type their Wifi details into the settings and then have full control of my Sonoff????

davidelang, 2018-01-06:
you should never run an AP on channel 8. On the 2.4 GHz band, the channels are 
too close together, so if you use channel 1, it overlaps channels 2 and 3, if 
you use channel 6 it overlaps channels 4,5,6,7,8 and if you use channel 11 it 
overlaps channels 9,10,11

if you use channels 1,6,11 they don't interfere with each other, and people in 
the area have three channels to use.

If you use channel 8, it clobbers channels 6,7,8,9,10 and as such, there are 
only two channels that can be used without interfering with each other.

davidelang, 2018-01-06:
you can turn off the AP mode, see the wifi settings

closed 2018-05-20 by stale[bot]