arendst/Tasmota #1528: Incorrect generation of MQTT FullTopic queue name

SGF-Lon, 2018-01-06:
I'm using the Master version of 5.10.0 and it has been working fine.  However, I am now making some changes to the naming conventions of my devices and just encountered the situation where 2 MQTT messages from the same device came out with different full topics.  The cmnd message is correct (i.e. it always has "Sonoff/" at the beginning - as specified.    However, the 'stat' message has lost the initial part of the Fulltopic as defined in the user-config (i.e. missing the "Sonoff/" at the beginning.

The MQTT messages are:
Received PUBLISH from openhab2 (d0, q0, r0, m0, 'Sonoff/cmnd/Sonoff_4238/POWER', ... (3 bytes))
Sending PUBLISH to Sonoff_4238 (d0, q0, r0, m0, 'Sonoff/cmnd/Sonoff_4238/POWER', ... (3 bytes))
Received PUBLISH from Sonoff_4238 (d0, q0, r0, m0, 'stat/Sonoff_4238/RESULT', ... (15 bytes))
Received PUBLISH from Sonoff_4238 (d0, q0, r0, m0, 'stat/Sonoff_4238/POWER', ... (3 bytes))

The web interface console displays INFORMATION as:
Hostname | Sonoff_4238
MQTT Client & Fallback Topic | Sonoff_4238
MQTT User | Sonoff_4238
MQTT Topic | Sonoff_4238
MQTT Group Topic | sonoffs
MQTT Full Topc | Sonoff/cmnd/Sonoff_4238/

However, when I change the MQTT configuration via the web console back to the previous values (see below) - the MQTT Fulltopic is back to normal.  The reverted MQTT parameters are:

The web interface console displays INFORMATION as:
Hostname | SW1_4238
MQTT Client & Fallback Topic | SW1_4238
MQTT User | Sonoff_4238
MQTT Topic | sonoffs
MQTT Full Topic | Sonoff/cmnd/Sonoff_4238/

The MQTT messages are:
Received PUBLISH from SW1_4238 (d0, q0, r0, m0, 'Sonoff/stat/Sonoff_4238/RESULT', ... (14 bytes))
PUBLISH to openhab2 (d0, q0, r0, m0, 'Sonoff/stat/Sonoff_4238/RESULT', ... (14 bytes))
Received PUBLISH from SW1_4238 (d0, q0, r0, m0, 'Sonoff/stat/Sonoff_4238/POWER', ... (2 bytes))
Sending PUBLISH to openhab2 (d0, q0, r0, m0, 'Sonoff/stat/Sonoff_4238/POWER', ... (2 bytes))

I think there must be a bug somewhere in generating the FullTopic when the Client & User & Hostname are the same.  But, there are actually the values I want to use.  I particularly would like to use a "prefix" followed by the 4 digit number derived from the MAC code (just like WIFI_HOSTNAME).  My preferred value for all 5 places (Wifi hostname, mqtt client, mqtt user, mqtt topic & the final part of the mqtt full topic) would be "Sonoff_4238" and I would really like to be able to specify that in the User_config.h by using something like the substitution characters used in creating the MQTT_CLIENT_ID & WIFI-HOSTNAME.  

Help - a bug & a change request?

SGF-Lon, 2018-01-22:
I have just discovered a second instance of the 'same' problem.  Below is the console log from one device where I sent a STATUS 2 command via MQTT @ both 17:03 & 17:10 and didn't receive the predicted MQTT full topic (i.e. the prefix was ignored - "stat/....").  However, when I sent a STATUS 2 command via the Web console @ 17:15 - then the Full topic was used correctly (i.e. "Sonoff/stat/..")

It can't be correct that the MQTT topic is different depending upon where the command comes from.

Is this a bug?

16:53:32 MQT: Sonoff/stat/Sonoff_3306/RESULT = {"POWER":"OFF"}
16:53:32 MQT: Sonoff/stat/Sonoff_3306/POWER = OFF
16:53:39 MQT: Sonoff/tele/Sonoff_3306/STATE = {"Time":"2018-01-........"}}
16:53:39 MQT: Sonoff/tele/Sonoff_3306/ENERGY = {"Time":"2018-01-.............}
17:02:00 MQT: Sonoff/tele/Sonoff_3306/UPTIME = {"Time":"2018-01-........."}}
17:03:39 MQT: Sonoff/tele/Sonoff_3306/STATE = {"Time":"2018-01-............"}}
17:03:39 MQT: Sonoff/tele/Sonoff_3306/ENERGY = {"Time":"2018-01-............."}}
17:03:47 MQT: stat/Sonoff_3306/STATUS2 = {"StatusFWR":{"Version":"5.10.0","BuildDateTime":"2018-01-08T13:14:11","Boot":31,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
17:08:39 MQT: Sonoff/tele/Sonoff_3306/STATE = {"Time":"2018-01-......................"}}
17:08:39 MQT: Sonoff/tele/Sonoff_3306/ENERGY = {"Time":"2018-01-............}
17:10:03 MQT: stat/Sonoff_3306/STATUS2 = {"StatusFWR":{"Version":"5.10.0","BuildDateTime":"2018-01-08T13:14:11","Boot":31,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
17:13:39 MQT: Sonoff/tele/Sonoff_3306/STATE = {"Time":"2018-01-......."}}
17:13:39 MQT: Sonoff/tele/Sonoff_3306/ENERGY = {"Time":"2018-01-............}
17:15:02 CMD: status 2
17:15:02 MQT: Sonoff/stat/Sonoff_3306/STATUS2 = {"StatusFWR":{"Version":"5.10.0","BuildDateTime":"2018-01-08T13:14:11","Boot":31,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}

SGF-Lon, 2018-02-02:
@arendst ..... Theo - any insight as to why this might be happening?  Thanks.

arendst, 2018-02-02:
I don't know the issue numbers but this has been answered several time. Look for using different names for fallbacktopic (= mqttclient) and topic.

SGF-Lon, 2018-02-03:
Theo.....thanks.  I've had a look and can't quite see the connection between the circumstances I've listed above and the setting of MQTT Topic and/or Fallback topic.  This is a situation where the MQTT TOPIC in the response to the exact same STATUS command is different depending upon whether the command is entered via the web server or via an MQTT message (see the log above).  Do you expect this behaviour?

AleksejsC, 2018-02-04:
Had the same issue. Changed MQQT Client name and now it is working as expected.

SGF-Lon, 2018-02-04:
Yes - I had to do the same thing.  But I really don't want to have to
change the naming convention of all of my Sonoff's to suit what looks to
be incorrect behaviour.
I don't believe that this can be correct behaviour.  Having to change
the MQTT Client name in order to get responses from identical MQTT
status requests albeit from different sources to go into the same MQTT
response queue doesn't sound right to me.
I think that somewhere in the code there is a check that two things are
equal (MQTT Client & Prefix??) and when they are - then there is an
incorrect assumption that this means that no prefix is needed.  Or
something along these lines.......
I've had a quick look at the code - but didn't see anything obvious.

Links:

  1. https://github.com/arendst/Sonoff-Tasmota/issues/1528#issuecomment-362899384
  2. https://github.com/

SGF-Lon, 2018-04-24:
I don't think that changing the client name should affect the functionality or the formatting of the MQTT queue name.  This is a situation where the MQTT TOPIC in the response to the exact same STATUS command is different depending upon whether the command is entered via the web server or via an MQTT message (see the log above).  @arendst does it seem right to you that the source of the command should affect the MQTT queue that receives the response?  I suspect that the somewhere the formatting of the MQTT queue name is being confused by the contents of the MQTT client name.  Thoughts?

arendst, 2018-04-26:
As said before. I have made the decision that if people loose their full topic information they can use the fallback topic to get to the device and reconfigure it. The fallback topic is the unique mqtt_client name.

If you opt for using the mqtt_client name as a topic you end up here.

closed 2018-06-25 by stale[bot]