arendst/Tasmota #1529: Sonoff 4CH pro relay behaviour

SGF-Lon, 2018-01-06:
I am about to receive a pair of Sonoff 4ch pro switches which have the NO/NC contacts exposed.  By cross-wiring 2 of the relays NO/NC contacts  together I can turn the device into an "intermediate light switch".  So, this would allow the device to act as an "intermediate light switch" for 2 pairs of lighting circuits.

I would like to be able to configure Tasmota to be able to receive one MQTT command and toggle one relay ON (e.g. configured to be relay 1) & the other relay OFF (e.g. configured to be relay 2).  The same principle would work for pairing relay 3 & 4 together so that the toggle together.

There is no need for combining status reporting or telemetry,  These can be handled independently via the 'normal' independent MQTT topics.  I think it's only disabling on/off commands on relay 2 and causing relay 2 to always be at the opposite value of relay 1 (even at initialisation).

Is this possible to do at the moment?  If not, could this be a change request so that this device can be used as an "intermediate light switch". 

I know that this can be achieved via a number of MQTT rules & commands.  However, I really don't want to have to rely upon MQTT to keep the wiring in the house straight.  Knowing the the toggling logic is in the firmware gives more security against accidents.

Thoughts?

SupraJames, 2018-01-07:
I have also given this some thought and was getting awfully confused with three way switching and intermediate switches and the like. Then it hit me - I'm trying to solve the wrong problem.

When I toggle an 'intermediate' switch, what do I WANT to happen?

I want the associated light to toggle, AND have accurate feedback to if the light is on or now.

So, my proposed solution is to replace the light switches, but IGNORE intermediate / 2 way / 3 way wiring - well, tie it together so it's irrelevant. You then have the light controlled only by 1 relay (the 'master') and the 'slave' or 'intermediate' switches are simply sending MQTT TOGGLE messages for the same effect.

SGF-Lon, 2018-01-07:
Supra....Yes, that is would be a complete & working solution - as long
as you are prepared to commit the entire lighting circuit to being
dependant upon the ESP8266's, tasmota & wifi.  I understand why some
people are prepared to make that commitment - in which case - your
solution is simple & workable.
However, if you aren't prepared to make that commitment - then you still
have a challenge.  How to add 'smart switches' into an existing 'legacy'
wiring circuit without becoming 100% dependent upon the 'smart
technology'.
I'm still in this latter camp - so for circuits with multiple switches
I'm looking into how to retain the traditional switches in most places
and only swap out (or augment) one switch.
 Hence, I'm looking to add the 4ch Pro to one circuit as an additional
 3rd intermediate into a lighting circuit that already has 4 switches
 (i.e. 2 regular & 2 intermediates )  already in it.
With more research, I might be able to use the backlog command (along
with opposite default starting positions) to achieve what I want.  But
I'm not 100% sure yet.
Getting accurate feedback on the circuit being live or not might be an
interesting challenge.  Haven't thought about that yet.
Thoughts from anyone?

Links:

  1. https://github.com/arendst/Sonoff-Tasmota/issues/1529#issuecomment-355818835
  2. https://github.com/

bensuffolk, 2018-01-09:
Since you have no and nc contacts on the relays, you don't actually need one relay on and one off, you can do that in the wiring. So all you need is to gang the relays together, I.e both on or both off at the same time. That might be more useful as a feature than the on / off pairing you asked for?

SGF-Lon, 2018-04-24:
@bensuffolk ...yes, that would work.  So, yes - pairing 2 relays in a 4ch Pro together would be a useful feature to be able to 'simulate' an intermediate relay.  @arendst - is there a way to do this in the current version?

arendst, 2018-04-26:
Is this an option: ``Backlog power1 on; power2 on``

closed 2018-07-10 by ascillato2