arendst/Tasmota #1532: Sonoff S22: TelePeriod not working as expected

tiberiuana, 2018-01-06:
From what I can tell, `TelePeriod` doesn't function correctly on Sonoff S22, configured as Sonoff TH. Instead of seconds, `TelePeriod` seems to represent a number of longer intervals.

e.g. For a `TelePeriod 10`, I get MQTT messages every ~1 minute. For `TelePeriod 20`, I get MQTT messages every ~2 minutes. I see this on two S22s units, configured identically.

For reference, I also have a POW with the same version of Tasmota, and `TelePeriod` works as expected.

Console log:
```
21:11:14 MQT: tga/t4/tele/INFO1 = {"Module":"Sonoff TH", "Version":"5.9.1", "FallbackTopic":"T4", "GroupTopic":"sonoffs"}
21:11:14 MQT: tga/t4/tele/INFO2 = {"WebServerMode":"Admin", "Hostname":"t4", "IPAddress":"10.71.107.117"}
21:11:14 MQT: tga/t4/tele/INFO3 = {"RestartReason":"Software/System restart"}

[...]

21:13:37 CMD: TelePeriod 10
21:13:37 MQT: tga/t4/stat/RESULT = {"TelePeriod":"10"}
21:13:37 MQT: tga/t4/tele/STATE = {"Time":"2018-01-06T21:13:37", "Uptime":0, "Vcc":3.152, "POWER":"OFF", "Wifi":{"AP":1, "SSId":"XXX", "RSSI":100, "APMac":"7A:8A:20:2A:13:57"}}
21:13:37 MQT: tga/t4/tele/SENSOR = {"Time":"2018-01-06T21:13:37", "AM2301":{"Temperature":18.7, "Humidity":69.9}, "TempUnit":"C"}
21:14:28 MQT: tga/t4/tele/STATE = {"Time":"2018-01-06T21:14:28", "Uptime":0, "Vcc":3.152, "POWER":"OFF", "Wifi":{"AP":1, "SSId":"XXX", "RSSI":100, "APMac":"7A:8A:20:2A:13:57"}}
21:14:28 MQT: tga/t4/tele/SENSOR = {"Time":"2018-01-06T21:14:28", "AM2301":{"Temperature":18.7, "Humidity":69.9}, "TempUnit":"C"}
21:15:19 MQT: tga/t4/tele/STATE = {"Time":"2018-01-06T21:15:19", "Uptime":0, "Vcc":3.152, "POWER":"OFF", "Wifi":{"AP":1, "SSId":"XXX", "RSSI":100, "APMac":"7A:8A:20:2A:13:57"}}
21:15:19 MQT: tga/t4/tele/SENSOR = {"Time":"2018-01-06T21:15:19", "AM2301":{"Temperature":18.7, "Humidity":69.9}, "TempUnit":"C"}

[...]

21:20:48 CMD: TelePeriod 20
21:20:48 MQT: tga/t4/stat/RESULT = {"TelePeriod":"20"}
21:20:50 MQT: tga/t4/tele/STATE = {"Time":"2018-01-06T21:20:50", "Uptime":0, "Vcc":3.141, "POWER":"OFF", "Wifi":{"AP":1, "SSId":"XXX", "RSSI":100, "APMac":"7A:8A:20:2A:13:57"}}
21:20:50 MQT: tga/t4/tele/SENSOR = {"Time":"2018-01-06T21:20:50", "AM2301":{"Temperature":18.8, "Humidity":70.0}, "TempUnit":"C"}
21:22:33 MQT: tga/t4/tele/STATE = {"Time":"2018-01-06T21:22:33", "Uptime":0, "Vcc":3.142, "POWER":"OFF", "Wifi":{"AP":1, "SSId":"XXX", "RSSI":100, "APMac":"7A:8A:20:2A:13:57"}}
21:22:33 MQT: tga/t4/tele/SENSOR = {"Time":"2018-01-06T21:22:33", "AM2301":{"Temperature":18.8, "Humidity":70.1}, "TempUnit":"C"}
21:24:13 MQT: tga/t4/tele/STATE = {"Time":"2018-01-06T21:24:13", "Uptime":0, "Vcc":3.152, "POWER":"OFF", "Wifi":{"AP":1, "SSId":"XXX", "RSSI":100, "APMac":"7A:8A:20:2A:13:57"}}
21:24:13 MQT: tga/t4/tele/SENSOR = {"Time":"2018-01-06T21:24:13", "AM2301":{"Temperature":18.8, "Humidity":70.1}, "TempUnit":"C"}
```

davidelang, 2018-01-06:
the other time I've heard of this problem was when the person had set deep sleep 
to try and save power.

tiberiuana, 2018-01-06:
Bingo, and in record time too! Thank you @davidelang :1st_place_medal: !

This was indeed caused by a `Sleep 250`. With `Sleep 0`, `TelePeriod` works as expected.

closed 2018-01-06 by tiberiuana