arendst/Tasmota #1533: Setting an 'absolute' SwitchTopic

SupraJames, 2018-01-06:
I have a T1 2CH. The first button has been left alone, the second has been redefined as a switch (mode 3).

SwitchTopic is set to 'james/test' however the mqtt topic gets expanded to cmnd/james/test/POWER2

I have two issues here

1. My 'target' device doesn't understand cmnd/
2. Again, it also doesn't understand POWER2 

How can I set things up so that pushing the button sends an MQTT message to a 'specific' topic which may not have anything to do with my Sonoff topic configuration?

SupraJames, 2018-01-07:
I've worked around this by having a script running on my machine which subscribes to the full topic, then publishes to the thing I want.

I'd still like to know if there is a native way to do this? I want my second switch to toggle a sonoff basic, which will barf at the /POWER2 part.

profesionalTI, 2018-01-07:
I think that  cmnd/james/test/POWER2 is wrong. You need to see in config MQTT the Name of your sonoff, in my case is sw01ee, for example  cmnd/sw01ee/POWER2 and parameter, see more command on:

 https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

Here you can find the correct parameter.

El 06-01-2018, a la(s) 17:39, SupraJames <notifications@github.com> escribió:

> cmnd/james/test/POWER2

SupraJames, 2018-01-07:
I am not sure we completely understand each other. My device is configured with a topic itself, and that all works fine, I can control the relays with the correct /cmnd/sonoff/power1 and power2 commands.

My query is around the publish message SENT by the sonoff when a switch is pressed. I want to publish a specific topic, but it gets expanded in a way I do not want.

To re-phrase:

How can I publish an MQTT message to an **arbitrary** topic when a switch is pressed?

profesionalTI, 2018-01-07:
Try with node-red Project I use it, it’s great

SupraJames, 2018-01-07:
Sure, I could do, but again I feel that's a workaround. I'm already using a python script to translate the MQTT into something else. Should be easy to send an arbitrary message straight from Tasmota itself. Anyone?

profesionalTI, 2018-01-07:
Node-red is more Easy, the sonoff work fine. You can received and send mqtt command, try is very good.

davidelang, 2018-01-07:
you set the switchtopic, and that will send a mqtt message for every switch 
action (use buttontopic for button actions

SupraJames, 2018-01-07:
David,

Sure, I am already using SwitchTopic. Please see my original post - basically, I set SwitchTopic to what I actually want, <topic>, but what I get is expanded to /cmnd/<topic>/POWER2 which my target device is not listening for.

davidelang, 2018-01-07:
On Sun, 7 Jan 2018, SupraJames wrote:

> Sure, I am already using SwitchTopic. Please see my original post - basically, I set SwitchTopic to what I actually want, <topic>, but what I get is expanded to /cmnd/<topic>/POWER2 which my target device is not listening for.

Then you need to change what your targt system is listening for.

switch2 will be power2, switch1 will be power1 (or just power if there is only 
one switch), switch3 will be power3, and switch4 will be power4

that's how you differentiate between different switches being pressed.

SupraJames, 2018-01-07:
I appreciate your comments but it does seem rather backward. I just want a button to send a message.

So, you say I need to change what the target is listening for. Let’s say I’m using that switch to toggle a remote Sonoff Basic, with it’s one relay. It’s going to get the command POWER2 TOGGLE. See the problem?

It’s entirely possible I am missing something here but I shouldn’t have to deploy something big like OpenHAB or HASS to do something this basic?

Perhaps the problem is that SwitchTopic is global and not set per switch? Hmm. Maybe that’s it. Would be a good enhancement (for me) to be able set an arbitrary topic and message per switch.

davidelang, 2018-01-07:
Tasmota doesn't do what you are trying to do.

Tasmota is designed to work with home automation software. As such, it's 
designed to send and receive messages to that system and let it decide what to 
do if there is a need to send messages to other devices.

Tasmota is not designed to let you have one button send a message to one device, 
and a different button send a message to a different device.

It just doesn't do that.

Many people have asked for per-switch settings like this, but it's not going to 
happen because it would take up too much memory and code space on the device to 
do this, and that would limit what can be done in the primary use case where you 
are working with a home automation system.

now, in your use case, you may be able to set the switch to switch1, in which 
case the command sent out will be cmnd/<switchtopic>/POWER

SupraJames, 2018-01-07:
Great explanation, thank you. It make sense when it’s stated clearly like that.

I will continue to use my python script now as a proxy until I’ve worked out where I want to go with the whole system.

closed 2018-01-07 by SupraJames