arendst/Tasmota #1537: GroupTopic for User defined GPIOs.

julianint, 2018-01-07:
Hi
I just expanded my Sonoffs with some additional relays. The Grouptopic only works for the original relay which is {1}. Is it possible to assign GroupTopic to the new relays? they are connected to Serial In {2} and Serial Out {3}.
Tks

davidelang, 2018-01-07:
grouptopic appies to the entire device, what is it you are trying to do that's 
not working?

julianint, 2018-01-07:
No, it is only working in the original relay. The new ones on the other pins (TX and RX) does not work with GroupTopic. 
I expanded the sonoff basic so now I can control 3 lights with just one sonoff basic, but the grouptopic aims to turn them all on and off at once, and so far I cannot get it working.

davidelang, 2018-01-07:
what do you mean "it doesn't work"

what command did you try?

julianint, 2018-01-12:
My GroupTopic is light_año, si the command is cmnd/light_all/power, and the payload is on or off. 
Sonoff modules respond but only for the default relay.  The new ones handled by TX and RX (dont remember GPIO pins), does not respond to the command.

davidelang, 2018-01-12:
On Thu, 11 Jan 2018, julianint wrote:

> My GroupTopic is light_año, si the command is cmnd/light_all/power, and the payload is on or off.
> Sonoff modules respond but only for the default relay.  The new ones handled by TX and RX (dont remember GPIO pins), does not respond to the command.

working as expected

power is the equivalent of power1

if you want to switch the second relay, you would issue the command 
cmnd/light_all/power2

for the third relay, power3, etc.

there is not one command that will switch all relays on a device

David Lang

julianint, 2018-01-12:
yes, each one separately works fine. There is no problem with that. 
What I would like to know is if there is any way to subscribe the grouptopic to the other relays so all of them respond to their single topics and the grouptopic.

davidelang, 2018-01-12:
They all do respond to the grouptopic

what you are not understanding is that there is no command to affect all relays 
at once, it doesn't matter if you are using grouptopic, or the regular topic

if you issue cmnd/grouptopic/power2 it will affect the second relay on all 
devices that have that grouptopic.

You issue has nothing to do with grouptopic, you want a new command "powerall" 
that will affect all relays on the device.

julianint, 2018-01-12:
Ok. Understand now. Tks for explaining that to me.

closed 2018-05-26 by stale[bot]