arendst/Tasmota #1543: Wemos D1 Mini (NodeMCU) - relays pulse on restart / power up

Thomas-ZA, 2018-01-07:
Hi there,

I'm hoping to use this firmware to control some gate motors, via some relays. My concern is that when I connect my NodeMCU to USB power, or when I tell the firmware to reboot the ESP8266 - the pins seem to get momentarily pulsed high... (which would end up opening gates randomly if the device is rebooted for whatever reason)

Does this mean I am using the wrong pins? or is there anything I can do to prevent this for some or all of the relays?

Currently testing with a 8 channel relay board, using D0 - D7

Any pointers would be greatly appreciated! :)

davidelang, 2018-01-07:
nope, there is no way to control what happens to the pins before tasmota starts 
running.

normally, this momentary pulse isn't enough to cause any problems, but there are 
use cases where it will.

you could try seting your relays to be off when the line is high, but if the 
relays have power before the ESP can boot up, you are still in trouble.

Thomas-ZA, 2018-01-07:
wow that was a quick response :)
perhaps there is something I can do with the circuitry to get around this - there must be something I can do, the sonoff guys manage to not get the false pulse initially with their hardware - so I must be missing something somewhere along the lines...

Much appreciated for the feedback though... I'll dig more in the direction of a hardware solution then.

TomaxMe, 2018-03-29:
Thomas did you find the solution? I'm stucked in the same point.

Thomas-ZA, 2018-03-29:
I haven't had time to look at this again yet - I still want to try adding
some pull-down resisters to the signal pins that go to the relays...

    Kind regards,

    Thomas Mason
  thomas@acunet.co.za
  Skype:  thomasmason
~~~~~~~~~~~~~~~~~~~~

On 29 March 2018 at 08:59, TomaxMe <notifications@github.com> wrote:

> Thomas did you find the solution? I'm stucked in the same point.
>
> —
> You are receiving this because you authored the thread.
> Reply to this email directly, view it on GitHub
> <https://github.com/arendst/Sonoff-Tasmota/issues/1543#issuecomment-377141393>,
> or mute the thread
> <https://github.com/notifications/unsubscribe-auth/Ahkfaunddfod6X5-MR7myIFFfsV8joT8ks5tjIZNgaJpZM4RVwvH>
> .
>

TomaxMe, 2018-03-29:
Thanks for quick feedback, T.

Smandurlo, 2018-08-25:
I have the same problem, everytime the D1 mini with the relay shield reboots or boots up, the relay is triggered for a short time, enough to open the gate I control with it. Not really a good thing as you might understand. Anyone got a solution? How to check if it happens before or after tasmota is started?

SirRedZ, 2018-11-26:
I think this is the result of residual code, and I have currently also got this "bug". It just started happening to me, although all day today this has not been a problem. I will attempt to reflash with a different BIN file to ensure user settings are all erased. I am trying to use Tasmota 6.3.0.