arendst/Tasmota #1545: Wemos D1 / NodeMCU and Domoticz - 8 relay control?

Thomas-ZA, 2018-01-07:
Hi there,

How do I define the IDX values for relay 5-8 if I am trying to control an 8 channel relay?

Running tasmota on a NodeMCU - and can toggle all the relays in the tasmota web interface, but I only see the option to define IDX numbers for relays 1-4. 

I've tried running the following in the console on the web interface:
 CMD: DomoticzIdx5=54
or
 CMD: DomoticzIdx5 54

but I get this response:
MQT: stat/sonoff/RESULT = {"Command":"Unknown"}

arendst, 2018-01-08:
Currently only idx 1-4 are supported

Thomas-ZA, 2018-01-08:
Thanks Theo,

where would be the right place to post a feature request - if there is support for 8 relays under the wemos - is it a big deal to add additional IDX for more control?

Much appreciated!

arendst, 2018-01-08:
It's a matter of using some more Settings flash space but the biggest challenge is to keep all settings in one array to ease programming. The neighboring settings would need to be moved which tend to be of a problem for people waiting to update their firmware regularly...

I just nail it to my list and if I see the light I might implement it some day ;-)

Thomas-ZA, 2018-01-08:
Thank you!

I appreciate the super-snappy feedback by the way 👍  :)

If you have any pointers as to where I could look to hack it in on my side? or if its quite involved - I can scale back my idea for now...

Misiu, 2018-01-22:
@Thomas-ZA  I'm looking for a way to use Wemos, relays and Tasmota together. Could You share You wiring?
@arendst looking forward for seeing this feature. 8 relays (or even more) would be awesome!

Thomas-ZA, 2018-01-23:
@Misiu - getting the relays to work should be fairly straight forward (depending on the relays you have). Easiest would be a 4-relay board (you can connect 8 to tasmota already - but depending on how you want to control them - you may be limited to being able to control 4 for now)

I connected the VCC of my relay board to the USB power (5v) pin of the NodeMCU, connected ground to ground, and then connected the signal pins of the board to different digital pins of the NodeMCU. 

Then in tasmota - you set the device type to NodeMCU, that allows you to configure each pin that you have connected to the wemos / NodeMCU as a relay out. In my specific case - I needed to define them as "iRelay" which is still a relay, just with the on / off logic inverted. (without this - Tasmota would say off when they were on, and on when they were off)

I hope this helps?

MacPit69, 2018-04-06:
i also wait for the 8-Channel-Support that would be very nice for my projects