arendst/Tasmota #1546: Wemos Switch and mqtt

kaareseras, 2018-01-07:
When setting a pin for "swith", in a wemod modue, the resulting mqtt message dont differ between swith1, swith2 etc.,

23:59:35 MQT: stat/sonofftest/RESULT = {"POWER":"ON"}
23:59:35 MQT: stat/sonofftest/POWER = ON
23:59:36 MQT: stat/sonofftest/RESULT = {"POWER":"OFF"}
23:59:36 MQT: stat/sonofftest/POWER = OFF

all swithes gives results as above. I need to be able to difrentiate between the swithces

davidelang, 2018-01-07:
how do you have them defined? how are they wired?

it also could be that you need to define additional relays (or switchtopic)

kaareseras, 2018-01-08:
[image]

Thank you for the fast reply

The configuration is as in the above link, but simply a pushbutton connected from the pin, connecting to gnd. the button works, but the resulting mqtt message, dont differ betwen the inputs.

kaareseras, 2018-01-08:
I had to set some relays for each push button, this is kind of waste of pins, but it works. is there a way to use the switch input, without setting relay?

davidelang, 2018-01-08:
try setting switchtopic, I think once you've done that it always identifies the 
switch

kaareseras, 2018-01-13:
Hi
I have now done some more test

current setup:
[image]

00:17:12 CMD: switchtopic
00:17:12 MQT: stat/sonoff/RESULT = {"SwitchTopic":"test"}
00:17:17 CMD: switchmode
00:17:17 MQT: stat/sonoff/RESULT = {"SwitchMode1":2}

when activating switch 1:
00:18:17 MQT: cmnd/test/POWER = ON
00:18:17 MQT: cmnd/test/POWER = OFF

when activating switch 2:
00:18:45 MQT: cmnd/test/POWER = TOGGLE

What is needed:
state/device1/switch1 = ON
state/device1/switch1 = OFF
state/device1/switch2 = ON
state/device1/switch2 = OFF

I cant figure out how to get induvidual messages for each switch, so I can identify with switch there has been activated. Only by having the extra relays and thus using up the pins??

kaareseras, 2018-01-13:
I now set the swithmode2 to 2, giving me the same on/off, but with the same topic, how can I set induvidual topic for each switch, or othervise see what switch is pressed

davidelang, 2018-01-14:
If you only have one relay defined, and do not set switchtopic, you get the 
behavior you are describing.

Either set a second relay pin, or define switchtopic (it seems like you really 
want switchtopic anyway)

boaschti, 2018-01-16:
You have to choose for any free pin -> RELAY1 and RELAY2 (if you use it or not). I think it's a bug but I get no answer in #1216 . Perhaps also interesting is: #1253 .

closed 2018-05-30 by stale[bot]