arendst/Tasmota #1548: Possible memory issues with T1 devices

SupraJames, 2018-01-08:
Hi have two Basic and two T1 2CH modules. The basics seem rock solid but the T1s are giving me trouble.

Symptoms vary but the most common one is the web server becoming unresponsive, but MQTT commands are OK, but then also MQTT might completely stop too.

I have configured syslog level 4 and I don't see any errors, just a lack of news ones, e.g:

```
Jan  8 12:20:24 sonoff-5027 ESP-MQT: tele/sonoff/STATE = {"Time":"2018-01-08T13:21:44","Uptime":1,"Vcc":3.146,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Tinkers-2.4G","RSSI":100,"APMac":"60:E3:27:AF:AB:37"}}
Jan  8 12:20:24 sonoff-5027 ESP-MQT: tele/sonoff/SENSOR = {"Time":"2018-01-08T13:21:44","Switch2":"ON"}
Jan  8 12:20:35 sonoff-5027 ESP-WIF: Checking connection...
Jan  8 12:20:35 sonoff-5027 ESP-WIF: Connected
Jan  8 12:20:55 sonoff-5027 ESP-WIF: Checking connection...
Jan  8 12:20:55 sonoff-5027 ESP-WIF: Connected
Jan  8 12:21:15 sonoff-5027 ESP-WIF: Checking connection...
Jan  8 12:21:15 sonoff-5027 ESP-WIF: Connected

```
Then nothing.

last time it happened the device was not able to work as a light switch, so suspect it was completely hung.

Web OTA firmware u/g also fails most of the time with the 'blocked loop' error.

I am running a custom build of 5.10.0h with just my own SSID in place. No TLS, No Emulation, no Dometicz.

I don't think it's a Wifi issue. RSSI is very high, even with a 10 meter separation (and two walls!)

```
Jan  8 11:36:40 sonoff-5027 ESP-MQT: tele/sonoff/STATE = {"Time":"2018-01-08T12:38:00","Uptime":0,"Vcc":3.149,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Tinkers-2.4G","RSSI":100,"APMac":"60:E3:27:AF:AB:37"}}
Jan  8 11:41:48 sonoff-5027 ESP-MQT: tele/sonoff/STATE = {"Time":"2018-01-08T12:43:08","Uptime":0,"Vcc":3.147,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Tinkers-2.4G","RSSI":100,"APMac":"60:E3:27:AF:AB:37"}}
Jan  8 11:46:48 sonoff-5027 ESP-MQT: tele/sonoff/STATE = {"Time":"2018-01-08T12:48:08","Uptime":0,"Vcc":3.147,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Tinkers-2.4G","RSSI":100,"APMac":"60:E3:27:AF:AB:37"}}
Jan  8 11:51:48 sonoff-5027 ESP-MQT: tele/sonoff/STATE = {"Time":"2018-01-08T12:53:08","Uptime":0,"Vcc":3.148,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Tinkers-2.4G","RSSI":98,"APMac":"60:E3:27:AF:AB:37"}}
Jan  8 11:56:49 sonoff-5027 ESP-MQT: tele/sonoff/STATE = {"Time":"2018-01-08T12:58:08","Uptime":0,"Vcc":3.149,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Tinkers-2.4G","RSSI":100,"APMac":"60:E3:27:AF:AB:37"}}
Jan  8 12:20:24 sonoff-5027 ESP-MQT: tele/sonoff/STATE = {"Time":"2018-01-08T13:21:44","Uptime":1,"Vcc":3.146,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Tinkers-2.4G","RSSI":100,"APMac":"60:E3:27:AF:AB:37"}}
```

Does anyone have a known good build which works well with these ESP8285 devices?

arendst, 2018-01-08:
\<Nice mode> Please use the pre-compiled binaries provided with the release as found on this site </Nice mode>

SupraJames, 2018-01-08:
Thank you for exnable nice mode for this newbie :)

Done, upgraded both and will monitor. Cheers

SupraJames, 2018-01-08:
OK, both have failed already unfortunately, about 20 minutes post-upgrade. They seem to be completely frozen and not responding to ping or even the touch buttons.

EDIT: Actually, touch is working to a fashion. Sometimes, about 15-20 seconds after touch, it's registered and the relay clicks over, so something is still alive. Still no ping or anything.

Complete syslog output for one of them follows.

```
Jan  8 16:56:05 bath-5027 ESP-MQT: stat/bath/UPGRADE = Successful. Restarting
Jan  8 16:56:27 bath-5027 ESP-RSL: stat/bath/RESULT = {"POWER1":"OFF"}
Jan  8 16:56:27 bath-5027 ESP-RSL: stat/bath/POWER1 = OFF
Jan  8 16:56:27 bath-5027 ESP-WIF: Checking connection...
Jan  8 16:56:27 bath-5027 ESP-WIF: Connected
Jan  8 16:56:27 bath-5027 ESP-DNS: Initialized
Jan  8 16:56:27 bath-5027 ESP-HTP: Web server active on bath-5027.local with IP address 192.168.100.153
Jan  8 16:56:28 bath-5027 ESP-CFG: Saved to flash at F9, Count 50, Bytes 1648
Jan  8 16:56:29 bath-5027 ESP-APP: (UTC) Mon Jan 08 16:57:47 2018
Jan  8 16:56:29 bath-5027 ESP-APP: (DST) Sun Mar 25 02:00:00 2018
Jan  8 16:56:29 bath-5027 ESP-APP: (STD) Sun Oct 28 03:00:00 2018
Jan  8 16:56:30 bath-5027 ESP-MQT: Attempting connection...
Jan  8 16:56:30 bath-5027 ESP-MQT: Connected
Jan  8 16:56:30 bath-5027 ESP-MQT: tele/bath/LWT = Online (retained)
Jan  8 16:56:30 bath-5027 ESP-MQT: cmnd/bath/POWER =
Jan  8 16:56:30 bath-5027 ESP-MQT: Subscribe to cmnd/bath/#
Jan  8 16:56:30 bath-5027 ESP-MQT: Subscribe to cmnd/sonoffs/#
Jan  8 16:56:30 bath-5027 ESP-MQT: Subscribe to cmnd/DVES_9893A3/#
Jan  8 16:56:30 bath-5027 ESP-MQT: tele/bath/INFO1 = {"Module":"Sonoff T1 2CH","Version":"5.11.1","FallbackTopic":"DVES_9893A3","GroupTopic":"sonoffs"}
Jan  8 16:56:30 bath-5027 ESP-MQT: tele/bath/INFO2 = {"WebServerMode":"Admin","Hostname":"bath-5027","IPAddress":"192.168.100.153"}
Jan  8 16:56:30 bath-5027 ESP-MQT: tele/bath/INFO3 = {"RestartReason":"Software/System restart"}
Jan  8 16:56:31 bath-5027 ESP-MQT: stat/bath/RESULT = {"POWER1":"OFF"}
Jan  8 16:56:31 bath-5027 ESP-MQT: stat/bath/POWER1 = OFF
Jan  8 16:56:31 bath-5027 ESP-MQT: stat/bath/RESULT = {"POWER2":"OFF"}
Jan  8 16:56:31 bath-5027 ESP-MQT: stat/bath/POWER2 = OFF
Jan  8 16:56:38 bath-5027 ESP-MQT: tele/bath/STATE = {"Time":"2018-01-08T17:57:57","Uptime":0,"Vcc":3.148,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Tinkers-2.4G","RSSI":98,"APMac":"60:E3:27:AF:AB:37"}}
Jan  8 16:56:48 bath-5027 ESP-WIF: Checking connection...
Jan  8 16:56:48 bath-5027 ESP-WIF: Connected
Jan  8 16:57:08 bath-5027 ESP-WIF: Checking connection...
Jan  8 16:57:08 bath-5027 ESP-WIF: Connected
Jan  8 16:57:28 bath-5027 ESP-WIF: Checking connection...
Jan  8 16:57:28 bath-5027 ESP-WIF: Connected
Jan  8 16:57:48 bath-5027 ESP-WIF: Checking connection...
Jan  8 16:57:48 bath-5027 ESP-WIF: Connected
Jan  8 16:58:08 bath-5027 ESP-WIF: Checking connection...
Jan  8 16:58:08 bath-5027 ESP-WIF: Connected
Jan  8 16:58:28 bath-5027 ESP-WIF: Checking connection...
Jan  8 16:58:28 bath-5027 ESP-WIF: Connected
Jan  8 16:58:48 bath-5027 ESP-WIF: Checking connection...
Jan  8 16:58:48 bath-5027 ESP-WIF: Connected
Jan  8 16:59:08 bath-5027 ESP-WIF: Checking connection...
Jan  8 16:59:08 bath-5027 ESP-WIF: Connected
Jan  8 16:59:28 bath-5027 ESP-WIF: Checking connection...
Jan  8 16:59:28 bath-5027 ESP-WIF: Connected
Jan  8 16:59:43 bath-5027 ESP-APP: (UTC) Mon Jan 08 17:01:01 2018
Jan  8 16:59:43 bath-5027 ESP-APP: (DST) Sun Mar 25 02:00:00 2018
Jan  8 16:59:43 bath-5027 ESP-APP: (STD) Sun Oct 28 03:00:00 2018
Jan  8 16:59:48 bath-5027 ESP-WIF: Checking connection...
Jan  8 16:59:48 bath-5027 ESP-WIF: Connected
Jan  8 17:00:07 bath-5027 ESP-HTP: Main Menu
Jan  8 17:00:08 bath-5027 ESP-WIF: Checking connection...
Jan  8 17:00:08 bath-5027 ESP-WIF: Connected
Jan  8 17:00:09 bath-5027 ESP-HTP: Information
Jan  8 17:00:11 bath-5027 ESP-HTP: Configuration
Jan  8 17:00:12 bath-5027 ESP-HTP: Configure Logging
Jan  8 17:00:16 bath-5027 ESP-HTP: Configuration
Jan  8 17:00:18 bath-5027 ESP-HTP: Configure Domoticz
Jan  8 17:00:20 bath-5027 ESP-HTP: Configuration
Jan  8 17:00:21 bath-5027 ESP-HTP: Configure WiFi
Jan  8 17:00:23 bath-5027 ESP-HTP: Configuration
Jan  8 17:00:28 bath-5027 ESP-WIF: Checking connection...
Jan  8 17:00:28 bath-5027 ESP-WIF: Connected
Jan  8 17:00:41 bath-5027 ESP-MQT: tele/bath/UPTIME = {"Time":"2018-01-08T18:02:00","Uptime":1}
[... 111 lines cut ...]
Jan  8 17:15:49 bath-5027 ESP-WIF: Connected
Jan  8 17:16:09 bath-5027 ESP-WIF: Checking connection...
Jan  8 17:16:09 bath-5027 ESP-WIF: Connected
Jan  8 17:16:11 bath-5027 ESP-APP: Button 2 multi-press 1
Jan  8 17:16:11 bath-5027 ESP-MQT: stat/bath/RESULT = {"POWER2":"ON"}
Jan  8 17:16:11 bath-5027 ESP-MQT: stat/bath/POWER2 = ON
Jan  8 17:16:12 bath-5027 ESP-CFG: Saved to flash at F4, Count 55, Bytes 1648
Jan  8 17:16:12 bath-5027 ESP-APP: Button 1 multi-press 1
Jan  8 17:16:13 bath-5027 ESP-MQT: stat/bath/RESULT = {"POWER1":"ON"}
Jan  8 17:16:13 bath-5027 ESP-MQT: stat/bath/POWER1 = ON
Jan  8 17:16:14 bath-5027 ESP-CFG: Saved to flash at FB, Count 56, Bytes 1648
Jan  8 17:16:17 bath-5027 ESP-APP: Button 2 multi-press 1
Jan  8 17:16:17 bath-5027 ESP-MQT: stat/bath/RESULT = {"POWER2":"OFF"}
Jan  8 17:16:17 bath-5027 ESP-MQT: stat/bath/POWER2 = OFF
Jan  8 17:16:17 bath-5027 ESP-APP: Button 1 multi-press 1
Jan  8 17:16:18 bath-5027 ESP-CFG: Saved to flash at FA, Count 57, Bytes 1648
Jan  8 17:16:18 bath-5027 ESP-MQT: stat/bath/RESULT = {"POWER1":"OFF"}
[... cut ...]
Jan  8 17:16:19 bath-5027 ESP-CFG: Saved to flash at F9, Count 58, Bytes 1648
Jan  8 17:16:29 bath-5027 ESP-WIF: Checking connection...
Jan  8 17:16:29 bath-5027 ESP-WIF: Connected
Jan  8 17:16:39 bath-5027 ESP-MQT: tele/bath/STATE = {"Time":"2018-01-08T18:17:58","Uptime":1,"Vcc":3.147,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Tinkers-2.4G","RSSI":100,"APMac":"60:E3:27:AF:AB:37"}}
Jan  8 17:16:49 bath-5027 ESP-WIF: Checking connection...
Jan  8 17:16:49 bath-5027 ESP-WIF: Connected
```

arendst, 2018-01-08:
Pls provide output of command status 0

SupraJames, 2018-01-08:
As requested:

```
19:19:20 CMD: status 0
19:19:20 MQT: stat/desk/STATUS = {"Status":{"Module":29,"FriendlyName":"Sonoff","Topic":"desk","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
19:19:20 MQT: stat/desk/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","Uptime":1,"Sleep":0,"BootCount":24,"SaveCount":65,"SaveAddress":"FB000"}}
19:19:20 MQT: stat/desk/STATUS2 = {"StatusFWR":{"Version":"5.11.1","BuildDateTime":"2018-01-07T21:51:09","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
19:19:20 MQT: stat/desk/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":4,"LogHost":"smartie","LogPort":514,"SSId1":"Tinkers-2.4G","SSId2":"Tinkers-2.4G","TelePeriod":300,"SetOption":"00000009"}}
19:19:20 MQT: stat/desk/STATUS4 = {"StatusMEM":{"ProgramSize":484,"Free":516,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
19:19:20 MQT: stat/desk/STATUS5 = {"StatusNET":{"Hostname":"desk-5027","IPAddress":"192.168.100.153","Gateway":"192.168.100.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.100.1","Mac":"60:01:94:98:93:A3","Webserver":2,"WifiConfig":2}}
19:19:20 MQT: stat/desk/STATUS6 = {"StatusMQT":{"MqttHost":"smartie","MqttPort":1883,"MqttClientMask":"DVES_X","MqttClient":"DVES_9893A3","MqttUser":"DVES_USER","MAX_PACKET_SIZE":512,"KEEPALIVE":15}}
19:19:20 MQT: stat/desk/STATUS7 = {"StatusTIM":{"UTC":"Mon Jan 08 18:19:20 2018","Local":"Mon Jan 08 19:19:20 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
19:19:20 MQT: stat/desk/STATUS10 = {"StatusSNS":{"Time":"2018-01-08T19:19:20"}}
19:19:20 MQT: stat/desk/STATUS11 = {"StatusSTS":{"Time":"2018-01-08T19:19:20","Uptime":1,"Vcc":3.148,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Tinkers-2.4G","RSSI":100,"APMac":"60:E3:27:AF:AB:37"}}}
```

arendst, 2018-01-08:
Looks fine although lowering syslog to 2 makes sense. Also do not use more than one webpage viewer as the webserver will certainly die within half an hour.

Look at status 0 again if response becomes slow and check heap.

Try to compile with esp8266/arduino v2.4.0 and see what happens.

SupraJames, 2018-01-08:
Thanks. Will see how stable things are overnight, and put one of them on core 2.4.0 tomorrow. I have one in the wall and one on the bench so troubleshooting is no problem.

SGF-Lon, 2018-01-08:
I am also having trouble with the webserver on the T1 2CH.  Same firmware (5.10.0 master) on all devices (POW, T1 2CH & S20) and all of the other webservers are working fine.  I can ping the T1 device & it responds.  I can send/receive MQTT and they work fine.  However, I can't access the webserver.  I've tried power re-cycling the T1 & everything comes back online (ping & MQTT).  But still no webserver.

Thoughts?

SupraJames, 2018-01-08:
Could be the same issue. Please monitor this thread and if your issue looks different create a new one.

@arendst I've just had another 'crash' however there was NO response, so I couldn't get Status at the time. it rebooted itself due to 'Blocked Loop' - I wasn't doing anything at the time.

I do have a cron job running Status 0 every 5 mins though. Here's the output showing the Status at 19.45, and recovery from a crash at 19.50 and another Status report.

I will probably go ahead and put the 2.4.0 based code on this one.

```
Jan  8 19:45:05 sonoff-6086 ESP-MQT: stat/sonoff/STATUS = {"Status":{"Module":29,"FriendlyName":"bathroom","Topic":"sonoff","ButtonTopic":"sonoff","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
Jan  8 19:45:05 sonoff-6086 ESP-MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://smartie:80/arduino/sonoff.ino.generic.bin","Uptime":1,"Sleep":0,"BootCount":12,"SaveCount":115,"SaveAddress":"F4000"}}
Jan  8 19:45:05 sonoff-6086 ESP-MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"5.11.1","BuildDateTime":"2018-01-07T21:51:09","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
Jan  8 19:45:05 sonoff-6086 ESP-MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":2,"LogHost":"smartie","LogPort":514,"SSId1":"Tinkers-2.4G","SSId2":"Tinkers-2.4G","TelePeriod":300,"SetOption":"00000009"}}
Jan  8 19:45:05 sonoff-6086 ESP-MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":484,"Free":516,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
Jan  8 19:45:05 sonoff-6086 ESP-MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-6086","IPAddress":"192.168.100.152","Gateway":"192.168.100.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.100.1","Mac":"60:01:94:98:97:C6","Webserver":2,"WifiConfig":3}}
Jan  8 19:45:05 sonoff-6086 ESP-MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"smartie","MqttPort":1883,"MqttClientMask":"SO_%06X","MqttClient":"SO_9897C6","MqttUser":"","MAX_PACKET_SIZE":512,"KEEPALIVE":15}}
Jan  8 19:45:05 sonoff-6086 ESP-MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Mon Jan 08 19:46:24 2018","Local":"Mon Jan 08 20:46:24 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99}}
Jan  8 19:45:05 sonoff-6086 ESP-MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-01-08T20:46:24"}}
Jan  8 19:45:05 sonoff-6086 ESP-MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-01-08T20:46:24","Uptime":1,"Vcc":3.141,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Tinkers-2.4G","RSSI":92,"APMac":"60:E3:27:AF:AB:37"}}}
Jan  8 19:50:45 sonoff-6086 ESP-WIF: Connected
Jan  8 19:50:45 sonoff-6086 ESP-DNS: Initialized
Jan  8 19:50:45 sonoff-6086 ESP-HTP: Web server active on sonoff-6086.local with IP address 192.168.100.152
Jan  8 19:50:47 sonoff-6086 ESP-MQT: Attempting connection...
Jan  8 19:50:47 sonoff-6086 ESP-MQT: Connected
Jan  8 19:50:47 sonoff-6086 ESP-MQT: tele/sonoff/LWT = Online (retained)
Jan  8 19:50:47 sonoff-6086 ESP-MQT: cmnd/sonoff/POWER =
Jan  8 19:50:47 sonoff-6086 ESP-MQT: tele/sonoff/INFO1 = {"Module":"Sonoff T1 2CH","Version":"5.11.1","FallbackTopic":"SO_9897C6","GroupTopic":"sonoffs"}
Jan  8 19:50:47 sonoff-6086 ESP-MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-6086","IPAddress":"192.168.100.152"}
Jan  8 19:50:47 sonoff-6086 ESP-MQT: tele/sonoff/INFO3 = {"RestartReason":"Blocked Loop"}
Jan  8 19:50:48 sonoff-6086 ESP-MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
Jan  8 19:50:48 sonoff-6086 ESP-MQT: stat/sonoff/POWER1 = OFF
Jan  8 19:50:48 sonoff-6086 ESP-MQT: stat/sonoff/RESULT = {"POWER2":"OFF"}
Jan  8 19:50:48 sonoff-6086 ESP-MQT: stat/sonoff/POWER2 = OFF
Jan  8 19:50:55 sonoff-6086 ESP-MQT: tele/sonoff/STATE = {"Time":"2018-01-08T20:52:14","Uptime":0,"Vcc":3.142,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Tinkers-2.4G","RSSI":100,"APMac":"60:E3:27:AF:AB:37"}}
Jan  8 19:51:31 sonoff-6086 ESP-MQT: stat/sonoff/STATUS = {"Status":{"Module":29,"FriendlyName":"bathroom","Topic":"sonoff","ButtonTopic":"sonoff","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
Jan  8 19:51:31 sonoff-6086 ESP-MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://smartie:80/arduino/sonoff.ino.generic.bin","Uptime":0,"Sleep":0,"BootCount":13,"SaveCount":116,"SaveAddress":"FB000"}}
Jan  8 19:51:31 sonoff-6086 ESP-MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"5.11.1","BuildDateTime":"2018-01-07T21:51:09","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
Jan  8 19:51:31 sonoff-6086 ESP-MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":2,"LogHost":"smartie","LogPort":514,"SSId1":"Tinkers-2.4G","SSId2":"Tinkers-2.4G","TelePeriod":300,"SetOption":"00000009"}}
Jan  8 19:51:31 sonoff-6086 ESP-MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":484,"Free":516,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
Jan  8 19:51:31 sonoff-6086 ESP-MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-6086","IPAddress":"192.168.100.152","Gateway":"192.168.100.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.100.1","Mac":"60:01:94:98:97:C6","Webserver":2,"WifiConfig":3}}
[... cut ...]
Jan  8 19:51:31 sonoff-6086 ESP-MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Mon Jan 08 19:52:50 2018","Local":"Mon Jan 08 20:52:50 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99}}
Jan  8 19:51:31 sonoff-6086 ESP-MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-01-08T20:52:50"}}
Jan  8 19:51:31 sonoff-6086 ESP-MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-01-08T20:52:50","Uptime":0,"Vcc":3.141,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Tinkers-2.4G","RSSI":100,"APMac":"60:E3:27:AF:AB:37"}}}
```

SupraJames, 2018-01-08:
Status of bench unit right now - MQTT OK but web server not responding. Heap looks OK?

```
Jan  8 20:27:20 desk-5027 ESP-MQT: stat/desk/STATUS = {"Status":{"Module":29,"FriendlyName":"Sonoff","Topic":"desk","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
Jan  8 20:27:20 desk-5027 ESP-MQT: stat/desk/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","Uptime":0,"Sleep":0,"BootCount":27,"SaveCount":112,"SaveAddress":"F4000"}}
Jan  8 20:27:20 desk-5027 ESP-MQT: stat/desk/STATUS2 = {"StatusFWR":{"Version":"5.11.1","BuildDateTime":"2018-01-07T21:51:09","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
Jan  8 20:27:20 desk-5027 ESP-MQT: stat/desk/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":2,"LogHost":"smartie","LogPort":514,"SSId1":"dd-wrt","SSId2":"Tinkers-2.4G","TelePeriod":300,"SetOption":"00000009"}}
Jan  8 20:27:20 desk-5027 ESP-MQT: stat/desk/STATUS4 = {"StatusMEM":{"ProgramSize":484,"Free":516,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
Jan  8 20:27:20 desk-5027 ESP-MQT: stat/desk/STATUS5 = {"StatusNET":{"Hostname":"desk-5027","IPAddress":"192.168.100.153","Gateway":"192.168.100.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.100.1","Mac":"60:01:94:98:93:A3","Webserver":2,"WifiConfig":2}}
Jan  8 20:27:20 desk-5027 ESP-MQT: stat/desk/STATUS6 = {"StatusMQT":{"MqttHost":"smartie","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9893A3","MqttUser":"DVES_USER","MAX_PACKET_SIZE":512,"KEEPALIVE":15}}
Jan  8 20:27:20 desk-5027 ESP-MQT: stat/desk/STATUS7 = {"StatusTIM":{"UTC":"Mon Jan 08 20:28:39 2018","Local":"Mon Jan 08 21:28:39 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
Jan  8 20:27:20 desk-5027 ESP-MQT: stat/desk/STATUS10 = {"StatusSNS":{"Time":"2018-01-08T21:28:39"}}
Jan  8 20:27:20 desk-5027 ESP-MQT: stat/desk/STATUS11 = {"StatusSTS":{"Time":"2018-01-08T21:28:39","Uptime":0,"Vcc":3.148,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":2,"SSId":"Tinkers-2.4G","RSSI":94,"APMac":"60:E3:27:AF:AB:37"}}}
```

SupraJames, 2018-01-09:
Update. The wall mounted module which I flashed with a core v2.4.0 compiled binary seems to have been stable now for 19 hours, which is great. I do think that I got into a mess yesterday as the more troubleshooting you do, the less stable they become. I was using syslog level 4, and querying status 0 with a cron job running every 5 minutes. Stopped all that, and seems better.

I hate to rock the boat by compiling another image but I do need to re-configure Button 2 as a Switch, so fingers crossed!

SupraJames, 2018-01-09:
Oh, damn, spoke too soon. Still getting odd behavior, but may be related to a different issue (ARP). Closing this one

closed 2018-01-09 by SupraJames