arendst/Tasmota #1550: Allow specific commands in webserver usermode

WoHinDu, 2018-01-08:
A lot of people have their sonoff accessible from the internet. That is fine for the MQTT but it is very insecure for the web interface. A common use case is to toggle the light, to toggle a relay or to read some sensor data over the web interface or more important they execute commands over the HTTP interface. 

As I understand the functionality of the user mode from the web server right, there is no way to set the web server back to the admin mode without any physical access to the device or secured access through MQTT. Unfortunately, the complete HTTP command line functionality is deactivated in the user mode. 

**I think it would make sense to allow certain commands in the user mode to read data and to toggle switches, but it must be forbidden to execute any admin command.** Especially to read and write wifi data. That would fix the biggest security whole: It would be no longer possible to read out of the sonoff the wifi password.

davidelang, 2018-01-08:
There are a lot of cases where someone can do a lot of damage by switching the 
device.

This firmware is not designed to be exposed directly to the hostile Internet. 
You should not do that.

Setup a VPN that you connect to that then gives you access to the devices inside 
your network.

But don't make the webserver available to everyone on the Internet, that is just 
begging for it to become part of a botnet.

WoHinDu, 2018-01-08:
Exactly, I completely agree with you, but if you read through the forums there are a many people out there who do this. Even though the web interface was never meant to be published to the internet a lot of people who do exactly this. So I think allowing some command calls in the user mode is one first step to make the web interface a bit more secure.

davidelang, 2018-01-08:
I think that just gives people a false sense of security.

closed 2018-05-23 by stale[bot]