arendst/Tasmota #1552: DTH22 gone 5.11.1

rene9900, 2018-01-09:
I had sensor DTH22 on my sonoff basic, set to DTH22 but since update it is renamed SI7021?
those two sensors are different ?

arendst, 2018-01-09:
See _releasenotes:  Add support for iTead SI7021 temperature and humidity sensor by consolidating DHT22 into AM2301 and using former DHT22 as SI7021 (#735)

If your sensors works fine just be happy with the name change.
If it doesn't work I suggest to select the AM2301 as that uses the same driver as the original DHT22 did.

rene9900, 2018-01-09:
okay thank you. i only discovered it because my node-red gauges stopped responding, i had a function node listening for the DTH22 command 👍

closed 2018-01-09 by arendst

ngalfas, 2018-01-16:
One side error is from MQTT JSON configurations for reading DHT22 values.
After the 5.11 upgrade these does not work, have to change the JSON formating code (ie on OpenHab - Error processing MQTT message) where DHT22 was used.