arendst/Tasmota #1555: Another wdt reset error on NodeMCU

eliott166, 2018-01-09:
Hi,

Similar to a lot of other closed issues, I'm getting a wdt reset on my NodeMCU.  None of the solutions on the other issues seem to fit my case as they were either for Sonoff devices or pin shorts.  In my case, when powered on, the NodeMCU connects successfully to the WiFi, then within a few seconds, I get the wdt reset.  The same error keeps looping.

Plan to attach DHT22 temp/humidity sensor, but I have nothing connected to any of the GPIO pins at the moment.  The board is powered via usb cable.  

Any suggestions?
Thanks

```
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_51B646, GroupTopic sonoffs) Version 5.11.1
00:00:00 WIF: Connecting to AP1 mySSID in mode 11N as sonoff-6216...
00:00:03 WIF: Connected

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_51B646, GroupTopic sonoffs) Version 5.11.1
00:00:00 WIF: Connecting to AP1 mySSID in mode 11N as sonoff-6216...
00:00:03 WIF: Connected

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld
```

fireport, 2018-01-12:
I have the same problem too with a wemos d1 r2

`00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_19493E, GroupTopic sonoffs) Version 5.11.1
00:00:00 WIF: Connecting to AP1 office in mode 11N as sonoff-2366...
00:00:03 WIF: Connected`

I compile the code by myself changing this setting
`#define WIFI_CONFIG_TOOL       WIFI_MANAGER  `

Danie10, 2018-01-20:
I'm getting the identical error with my Sonoff POW and I have also tried the developer version. Have also tried two different Sonoff's. I tested the 2nd Sonoff with its standard firmware before I flashed it and it was fine. Only odd thing I'm noting, is I'm not seeing the blue LED flash briefly at the conclusion of the flashing.

00:00:00 Project sonoff1 Sonoff1 (Topic sonoff1, Fallback DVES_4E9F48, GroupTopic sonoffs) Version 5.11.1d
00:00:00 WIF: Connecting to AP1 mySSID in mode 11N as sonoff1-8008...
00:00:03 WIF: Connected

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld

crewmate1, 2018-01-24:
I have the same issue where it connects to WI-FI and then resets. I changed the config file to not auto connect to my network and start in webserver mode. It booted and launched the webserver. Once I configured the model and the network it started the reboot loops again.

arendst, 2018-01-24:
I see the reboot watchdog appear but only if I compile with:

- Esp8266/Arduino v2.4.0
- Arduino IDE
- Arduino IDE - Tool - LwIP Variant: V2 Prebuilt (MSS=536)

So I've updated the wiki - Arduino IDE part to select LwIP Variant: 1.4 Prebuilt which works just fine.

davidelang, 2018-01-24:
check that there is not a retained reset message in your mqtt server

set logging to 'more detail' and see what is happening as it boots, connects, 
and restarts.

ssjha, 2018-02-22:
I was also facing this issue for my Wemos modules. However setting the below option for Wemos worked from me. Thanks Theo.
_Arduino IDE - Tool - LwIP Variant: V2 Prebuilt (MSS=536)_
However i still get random wdt reset in Wemos modules. i have to reflash them again to get it working which is a big reliability issue. However i am not facing issues with my Sonoff and Nodemcu modules.