arendst/Tasmota #1560: Won't actually work

simo26246, 2018-01-10:
I have been using these switches (sonoffpow) for a while now (previously running version 5.6.1 without issue).

I decided that I'd reflash (using Arduino IDE) them and update the version, because I needed to change the wifi settings. So I did all the correct steps, and changed the settings (previously uploaded using DIO, 1M 64K SPIFFS, now use DOUT 1M No SPIFFS).

Firmware uploads fine, but then get this error - 

```
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_075D3E, GroupTopic sonoffs) Version 5.11.1
00:00:00 WIF: Connecting to AP1 Rogers in mode 11N as sonoff-7486...
00:00:03 WIF: Connected

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

```
and the device won't actually connect to the wifi network, or work.

When I plug it into the mains power, the blue wifi light blinks at approx 8 second intervals. I have no idea what that means, does anyone have any idea how I can fix this? 

This has now happened to 2 switches, I'm scared to do the rest for fear of bricking them.

Any help would be appreciated

gajotnt, 2018-01-10:
Make sure the wifi settings are correct and the device can get signal where it is. 
Sometimes this happens to me, but its a sonoff that is somewere the wifi signal is very weak.

simo26246, 2018-01-11:
Thanks for the reply gajotnt, the switch is 1.5m away from the router, and can confirm that the details are correct. 

By way of update, I've used esptool to erase the flash on both of them, then reflashed them, and get the same error. I'm assuming that this means I've bricked my devices?

davidelang, 2018-01-11:
the only way to brick an ESP is to apply too much voltage to it.

assuming you aren't dealing with a POW, what shows up on the serial port as it's 
booting up?

simo26246, 2018-01-11:
I am dealing with a POW.

I also realise now that I may have plugged 3.3v to the TX port on the POW...

simo26246, 2018-01-12:
By way of update, I've now gone back to 5.6.1, reflashed and have no issues. This makes me think it's something to do with the way I've setup the 5.11.0 firmware?

geekypenguin, 2018-01-13:
@simo26246  Please mark the issue as closed if you are no longer facing problems