arendst/Tasmota #1563: Non 192 Addressing

boilermanc, 2018-01-10:
Hi all, first of all, I just got a new batch of the basics in. How do you tell what is running on them without connecting them to ewlink? 

Second, i am using the exe file to install. Everything seems to work until you get to the part where it is waiting on the FinalStage connection. It just hangs at that point. Its looking for the 192.168 address but i run a 10. addressing. Is that an issue? If not, what else might be making it hang? 

thanks!
Clint

SupraJames, 2018-01-10:
I don’t know of a way of getting the version without going through the EWeLink setup. Your own network addressing doesn’t make a difference so don’t worry about that.

Out of 4 modules I’ve only ever had one work with the SonOTA script, so I just end up hard flashing.

webtop, 2018-01-11:
To see what is running on them you'll need to connect to the serial interface and see what the output is. That being said, I don't know if serial debug is on initially, so that may be an issue.

What exe are you talking about, and to install what?

reloxx13, 2018-01-12:
SonOTA.exe, no need to install python anymore, all is includes in the .exe.
Made with pyinstaller.

closed 2018-05-26 by stale[bot]