arendst/Tasmota #1565: Sonoff RF Bridge not joining MQTT (rc-2) - TAS menu only accessible via WiFi

RKF-FKR, 2018-01-10:
Well this one has me completely stumped. 

I have a number of devices working beautifully with my Pi MQTT server.

This little fella will not connect to the server.  
Same user and pass.
Different topic / group / 
Different Friendly name

The weird thing is that the Bridge menus can only be accessed my my wifi devices.  If I try to open the sonoff menu from a computer that has hard wired ethernet connections, the Tasmota page won't load.
My iPhone and a wifi only PC are was able to access the device menu.

I'm assuming the -rc-2 error for the MQTT and the fact that I can't access the device menu via Ethernet are related.

Am I missing something ultra basic here?

Any help would be greatly appreciated.
Rod

RKF-FKR, 2018-01-17:
All sorted - please close this issue - Sorry to bother

closed 2018-05-31 by stale[bot]

WagnerBeccard, 2018-06-26:
Jea.. But now i got this problem.. How you fixed it?
(but I cant even connect to the webserver)

RKF-FKR, 2018-06-26:
I bought a new sonoff bridge and everything flashed as expected. I discarded the other bridge. 
But I have had this problem on other wemos D1 mini also. In this case just choosing the exact settings in Arduino fixed the problem.