arendst/Tasmota #1566: Real Push button Functionality

Swarley321, 2018-01-10:
Hello,

Is it possible to switch on the relay of the sonoff touch while holding the touch button and turn of the relay after releasing the button? So only on when button pressed. I would like to control may window shades and those move when power is applied otherwise not. I do not like the push once to move and then push another time to stop the shades. I went trough all tasmota switch-modes and none seems to be the right one. Is there maybe a special derivative?

davidelang, 2018-01-10:
I think the 'follow' mode does what you want.

be aware, switchmodes don't apply to buttons, only to switches, so you would 
need to do a custom config (either in the source or with the wemos module type) 
to define the built-in buttons to be switches instead.

Swarley321, 2018-01-12:
Hello,

thanks for your answer. What I mean is ON when button is pushed and OFF when released. Just as shown in this video (after 25 sec.) https://www.youtube.com/watch?v=T2n0GvjESUM
This would be a nice feature for shade control. It seems that I really have to dive deep into the Firmware in order to get this working.

davidelang, 2018-01-12:
this is what switchmode = follow does. when the switch transitions from off to 
on, an 'on' event happens, when the switch transitions from on to off an 'off' 
event happens (event being either local relay change, mqtt message, etc)

Swarley321, 2018-01-13:
I checked again what switchmode follow does, and in a way it is similar what I mean but not quite there. 
From what I have seen the switchmode 1 or follow does this:
->You put the finger onto the touch surface and then release the finger from it
->Status ON is sent and relay is switched to ON
->You repeat step1 
->Status OFF is sent and relay is switched to OFF

What I want is this:
->You put the finger onto the touch surface
->Status ON is sent  and relay is switched to ON
->You release the finger from the touch surface 
->Status OFF is sent  and relay is switched to OFF

I hope that I brought across what I mean. I wonder if some hardware adjustments would also be needed.

BR Asmir

davidelang, 2018-01-13:
On Sat, 13 Jan 2018, Swarley321 wrote:

> I checked again what switchmode follow does, and in a way it is similar what I mean but not quite there.
> From what I have seen the switchmode 1 or follow does this:
> ->You put the finger onto the [...]

that is what toggle does

> What I want is this:
> ->You put the finger onto the touch surface
> ->Status ON is sent  and relay is switched to ON
> ->You release the finger from the touch surface
> ->Status OFF is sent  and relay is switched to OFF

this is exactly what follow does.

note, you can't do this with the built-in button, only with a switch input

ageorgios, 2018-01-16:
I think the same is dicussed here https://github.com/arendst/Sonoff-Tasmota/issues/1424

TarikSL, 2018-01-18:
Hi,
I use for shades the PulseTime 110. this leaves the motor running 10 seconds and turns off the relay. 
If using 2 gang, make sure to set the interlock option to on in order not to move up and down the motor at the same time

Swarley321, 2018-01-18:
Thanks for your input guys! 

I have to admit I am a little confused, davidelang so I have to change in the source code that the touch button of my Sonoff T1 is recognized as a switch, and then I change in the console of the tasmota firmware the switchmode to follow and I will get my above mentioned functionality, right? If yes, can you tell me if there is an instruction for the code change somewhere?

ageorgios thanks for your suggestion. I read in you link mainly that it is hard to program the T1. It was hard for me to flash the device the first time too, but then solved it and documented it in issue #1439.

TarikSL I used your input to set the interlock, and it works great. I only need to figure out how to turn on the relay only when i touch the button surface and I am done.

TarikSL, 2018-01-19:
From my little experience, you don’t need / have to hold touching the button -otherwise you’ll reset the sonoff configuration -, I use PulseTime to make the relays on for a few seconds to open entirely the shades.  My shades stops opening when it arrives at the end of the course. 
See also the option SwitchMode

Swarley321, 2018-01-21:
@TarikSL I see thank you very much. I changed option SwitchMode to 2 hoping that is does the pushbutton properly but it is basically the same behaviour as SwitchMode 1. @davidelang states that it should be exactly what I need but it is not different that SwitchMode 1 on my T13CH sonoff touch. I will probably give it up when it pushing the buttons longer initiates a reset of the switch. Btw. can this reset functionality be turned off, because I am afraid that kids will someday play around with the switch and reset it unintentionally

ageorgios, 2018-02-18:
Tell us your experience!
What setting did you use eventually?
I am planning a Sonoff T1 2CH for the same use with these options:

setoption13 1; setoption14 1; setoption0 0; setoption32 100; PowerOnState 0; timezone 2; setoption3 0; setoption1 1; ledstate 0;

and when triggering I use:
backlog pulsetime1 12; power1 on; delay 12; pulsetime 120

Mathyass, 2018-03-24:
I read all comments but no success. First i look on switchmode - was 0 so if i understand well its TOGGLE. touch the switch ON releas nothing and touch again OFF ...asi i see in console
23:30:28 CMD: switchmode1
23:30:28 MQT: stat/sonoff/RESULT = {"SwitchMode1":0}
23:30:34 MQT: stat/sonoff/RESULT = {"POWER":"ON"}
23:30:34 MQT: stat/sonoff/POWER = ON
23:30:35 MQT: stat/sonoff/RESULT = {"POWER":"OFF"}
23:30:35 MQT: stat/sonoff/POWER = OFF

so i tried switchmode1 1 - FOLLOW what is what i want .... touch switch -ON release finger and relay will be OFF but the behaviour is the same ...
23:32:12 CMD: Switchmode 1 1
23:32:12 MQT: stat/sonoff/RESULT = {"SwitchMode1":1}
23:32:19 MQT: stat/sonoff/RESULT = {"POWER":"ON"}
23:32:19 MQT: stat/sonoff/POWER = ON
23:32:21 MQT: stat/sonoff/RESULT = {"POWER":"OFF"}
23:32:21 MQT: stat/sonoff/POWER = OFF

I read also about type Button and Type SWITCH i tried to setup in web configuration both GPIO to switch but maybe it is not good way ...

spending hours how to solve it and googling a lot but no success. Can anyboy help me ? I think it will be very easy but i cant see it