arendst/Tasmota #1567: PIR Sensor

wirelesssolution, 2018-01-11:
HI , Any idea 
 How to user PIR Sensor for Sonoff Basic ?

davidelang, 2018-01-11:
yes we do.

if you want a more detailed answer, you need to ask a more detailed question

wirelesssolution, 2018-01-11:
I connect this type PIR sensor to Sonoff Baisc on GPI14 
https://i1.wp.com/g01.a.alicdn.com/kf/HTB1G1trKFXXXXXjXXXXq6xXFXXXe/SR501-HC-SR501-Adjust-IR-Pyroelectric-Infrared-font-b-PIR-b-font-module-Motion-Sensor-Detector.jpg_220x220.jpg?w=640
set as O9Switch1
it alway switch ON and OFF

davidelang, 2018-01-11:
On Wed, 10 Jan 2018, Wirelesssolution wrote:

> I connect this type PIR sensor to Sonoff Baisc on GPI14
> https://i1.wp.com/g01.a.alicdn.com/kf/HTB1G1trKFXXXXXjXXXXq6xXFXXXe/SR501-HC-SR501-Adjust-IR-Pyroelectric-Infrared-font-b-PIR-b-font-module-Motion-Sensor-Detector.jpg_220x220.jpg?w=640
> set as O9Switch1

how do you have it wired?

> it alway switch ON and OFF

what do you mean by this?

wirelesssolution, 2018-01-11:
I use 
SONOFF 3V -> PIR  VCC
SONOFF GND - > PIR GND
SONOFF GPIO14 -> PIR OUTPUT PIN
[image]

This is log file , with out movement 

07:52:13 MQT: stat/Kitchen_Power3/RESULT = {"POWER":"OFF"}
07:52:13 MQT: stat/Kitchen_Power3/POWER = OFF
07:52:16 MQT: stat/Kitchen_Power3/RESULT = {"POWER":"ON"}
07:52:16 MQT: stat/Kitchen_Power3/POWER = ON
07:52:18 MQT: stat/Kitchen_Power3/RESULT = {"POWER":"OFF"}
07:52:18 MQT: stat/Kitchen_Power3/POWER = OFF
07:52:52 MQT: stat/Kitchen_Power3/RESULT = {"POWER":"ON"}
07:52:52 MQT: stat/Kitchen_Power3/POWER = ON
07:53:27 MQT: stat/Kitchen_Power3/RESULT = {"POWER":"OFF"}
07:53:27 MQT: stat/Kitchen_Power3/POWER = OFF
07:53:30 MQT: stat/Kitchen_Power3/RESULT = {"POWER":"ON"}
07:53:30 MQT: stat/Kitchen_Power3/POWER = ON
07:53:55 MQT: stat/Kitchen_Power3/RESULT = {"POWER":"OFF"}
07:53:55 MQT: stat/Kitchen_Power3/POWER = OFF
07:53:57 MQT: stat/Kitchen_Power3/RESULT = {"POWER":"ON"}
07:53:57 MQT: stat/Kitchen_Power3/POWER = ON

davidelang, 2018-01-11:
you need to add a pull up resistor between the 3v and the output pin (nd make 
sure the PIR will run on 3.3v and not require 5v)

David Lang

wirelesssolution, 2018-01-11:
Thank you very much , I will try and update result :D

mkh595, 2018-01-11:
This sensor may not work properly with 3.3V

Jason2866, 2018-01-22:
In my case i needed a pull down resistor. PIR connected to Wemos mini.
Without a resistor sometimes it switched on. With a pull up resistor, it immediatly switched to on, after the on time of the Pir was over. Probably the output goes in tristate and then the pull up resistor did his job....

daballiemo, 2018-01-25:
Just if anybody is wondering about the 3,3 V:
http://techgurka.blogspot.nl/2013/05/cheap-pyroelectric-infrared-pir-motion.html

jaburges, 2018-02-08:
bought some of these  [](http://a.co/i9eIPUj
)
been a long while since i've dusted off the soldering iron (wasnt too shabby back then though).

It runs perfectly well on 3.3V and the best bit, it seems to see through a lot of objects. With some basic testing i can put it behind my monitor or curtain and it can still detect presence.

However annoyingly the output is inverted (no biggy as i can swap the payloads in homeassistant)

My question was around switchtopic - i really dont understand that yet. I wanted to use 'Switch2' on GPIO14
Then decouple it from the physical relay. Can you still report ON/OFF state of a virtual relay?
there is no need to keep triggering the relay

rathgar, 2018-02-13:
That sensor does work reliably if wired via @daballiemo 's 3.3v link (tested on a basic). Unfortunately it only seems to be able to trigger the internal sonoff switch and then a couple of seconds later toggle it back. It doesn't act as a sensor and is not visible nor does it broadcast any state changes on MQTT. 

I dug around in Tasmota source to see if I could work out how to write some sensor code for it but couldn't figure it out. For now I'm using automation logic to switch lights on/off in response the the toggling of the switch but it seems a huge waste of a sonoff. I think this is what @jaburges (and others) would be after. 

Anyone with any sensor knowledge able to point me in the right direction? Would be cleaner to have the PIR act as a sensor rather than a relay trigger.

johoberm, 2018-02-13:
@jarburges,

Yes this is possible. 
Set GPIO 14 (Sonoff Basic free Port) to  Switch2.
As Long PIR is ON, Switch2 has Status ON
when PIR is OFF, Switch 2 has Status OFF.
The Sonoff Relais isnt switched....

rathgar, 2018-02-13:
@johoberm is right. However, telemetry is only published every `TELE_PERIOD` seconds (default 5mins) rather than when motion is detected. I would love to work out how to use it as a sensor, rather than a trigger.

johoberm, 2018-02-13:
@rathgar, in realife it just works :-) I have this solution in my Homeoffice. If i went in the PIR detects me
and Switches the light and my computer on. I do this via a rule in OpenHab. The trigger is a mqqt message with the info JSON encoded.
One rules file and the PIR 
[Pir.items.txt](https://github.com/arendst/Sonoff-Tasmota/files/1720728/Pir.items.txt)
items file as example
[Pir.items.txt](https://github.com/arendst/Sonoff-Tasmota/files/1720729/Pir.items.txt)
[Pir.rules.txt](https://github.com/arendst/Sonoff-Tasmota/files/1720730/Pir.rules.txt)

Jason2866, 2018-02-13:
Here is the console output (different PC, different account on Github Jason2866 = johoberm)
Triggered via PIR
17:50:48 MQT: stat/sonoff-87786C/POWER = ON
17:50:49 MQT: stat/sonoff-87786C/RESULT = {"POWER":"OFF"}
17:50:49 MQT: stat/sonoff-87786C/POWER = OFF

daballiemo, 2018-02-13:
If you select GPIO14 sensor = 10 Switch2 (webinterface -> Configuration-> Module) 
you will see following on MQTT that you can use to trigger:
18:03:25 MQT: home/esp8266/tasmota/sonoff_BC4E0C/cmnd/POWER2 = ON
when the PIR is triggered.

Jason2866, 2018-02-13:
Teamwork😎
Now it should work as you wanted

rathgar, 2018-02-13:
Thanks for the examples @johoberm, that's pretty much mow I solved it on HA - a rule that turns on other things when the sonoff with the pir gets switched on. 

The better solution would be to have the pir just *report* that it detected motion via mqtt. That would leave the decision as to whether to turn on the device connected up to the central logic of OopenHab/Hassio/etc. I've not worked out if this is impossible or not... 🤓

Jason2866, 2018-02-13:
@rathgar My setup does that. I have the PIR connected to a Wemos mini. It only detects motion
and over OpenHab it does the Magic. The Wemos has connected a temperature / humidity sensor too.
I use that for controlling my heating system. And it switches off my heating if i open the window and
turn it on again when its closed. Thats all be can be done with monitoring and rules with more or less complex logic. I find it awesome what is possible. It is like a drug. If things get rolling, you have next idea what could be done. And that all for that little money.

rathgar, 2018-02-13:
@Jason2866 That makes sense then, I was working solely within a sonoff/tasmota world, blissfully unaware of wemos. You are right, it's very addictive. When my soldering gets more confident I have some LED strips to flash :)

Gastondamian, 2018-08-07:
go to tasmota console and write:

switchmode2 1

aihysp, 2018-09-19:
Hi, I have a strange issue for me ... but I guess its because I know nothing about electronics :)
using the little PIR sensor with 3 pins, and no knobs 
when I do bench testing with a big breadboard I just connect the small PIR sensor to the sonoff
GPIO 14 >middle pin(labeled 0)
GND >1 pin (labeled  1 wired no?)
3.3 >  +

in gui i define GPIO14 as switch2
when i  look at console i get nice results 

cmnd/pir/POWER2 = TOGGLE 
each time there is an event it's working very cool on the breadboard!

but when I take it out wired things start happening, things like I don't get results, or I get too many false ones

maybe I need some resistor, as I am noob I have no idea what values 
i guess the length of the cable has an effect, or maybe the breadboard has res
anyway, if someone has an idea how to use this small PIR sensor I will glad to hear!

gerardovf, 2018-09-25:
Hi, I have another issue with GPIO14 on a Sonoff Basic when using a PIR.
Everything works as expected BUT if the Sonoff is not able to connect to the MQTT broker, then when the PIR is triggered the relay is operated!

mirko, 2018-12-02:
I do also experience - whatever SwitchX I use for GPIO14 - the sensor immediately switches on the internal relay, the the sonoff is not connected.

ascillato, 2018-12-03:
Please, use rules for PIR. Switchtopic is for switches.

olinj, 2019-01-09:
I am new to this and have been digging through youtube to try to figure out how to set up a sonoff basic that will have a PIR sensor and connected to a regular light switch.  I have hassio and tasmota flashed.  I have configured the sonof (named "pantry") and it can be turned on and off through my home assistant.  
I have not added any other code to add the PIR sensor because I dont know which pin to use....GPIO 14, tx, or rx.  I know to use 3.3v and gnd.  I also believe the switch needs to be wired to gpio 14 and gnd to work.  Can anyone direct me to a page that has the recomended pin and coding for this project?  
I want the motion sensor to turn on the light when someone walks into the pantry and shut off in 30 seconds without motion.  I also want to do this same project on another switch but also add a temp and humidity sensor, same set up connected to a switch (toggle not push button switch)
thanks in advance

betyar, 2019-02-14:
I have an AM312 configured as Switch1 for a dummy Relay1 (I am using 18 Generic Configuration). The PIR keeps switching on and then immediately off every time it's triggered. Does anyone have a solution for this?

Here is an example from my console:

22:17:39 MQT: tele/sonoff-attic/STATE = {"Time":"2019-02-14T22:17:39","Uptime":"0T00:05:20","Vcc":3.104,"POWER":"OFF","Wifi":{"AP":1,"SSId":"betyar0","RSSI":46,"APMac":"00:23:CD:11:B1:9C"}}
22:17:40 MQT: tele/sonoff-attic/SENSOR = {"Time":"2019-02-14T22:17:39","Switch1":"OFF"}
22:18:37 MQT: stat/sonoff-attic/RESULT = {"POWER":"ON"}
22:18:37 MQT: stat/sonoff-attic/POWER = ON
22:18:37 MQT: stat/sonoff-attic/RESULT = {"POWER":"OFF"}
22:18:37 MQT: stat/sonoff-attic/POWER = OFF
22:18:52 MQT: stat/sonoff-attic/RESULT = {"POWER":"ON"}
22:18:52 MQT: stat/sonoff-attic/POWER = ON
22:18:52 MQT: stat/sonoff-attic/RESULT = {"POWER":"OFF"}
22:18:52 MQT: stat/sonoff-attic/POWER = OFF
22:19:15 MQT: stat/sonoff-attic/RESULT = {"POWER":"ON"}
22:19:15 MQT: stat/sonoff-attic/POWER = ON
22:19:15 MQT: stat/sonoff-attic/RESULT = {"POWER":"OFF"}
22:19:15 MQT: stat/sonoff-attic/POWER = OFF
22:19:38 MQT: stat/sonoff-attic/RESULT = {"POWER":"ON"}
22:19:38 MQT: stat/sonoff-attic/POWER = ON
22:19:38 MQT: stat/sonoff-attic/RESULT = {"POWER":"OFF"}
22:19:38 MQT: stat/sonoff-attic/POWER = OFF

Because the switching is so fast the web page doesn't show the change between ON and OFF. Also, if I manually toggle the switch, then the situation is the reverse, that is, when the PIR is triggered the console shows OFF immediately followed by ON.

Any help is much appreciated.

Jason2866, 2019-02-14:
Tried a pull up or pull down resistor as described in this thread?

betyar, 2019-02-15:
Yes, I tried both. The only difference is that it doesn't show when the PIR was triggered, but the end result is the same in both cases. Below is the log from my console showing the results when there was a pull down, pull up, and when there was no resistor at all.

Result when GPIO is pulled LOW

00:00:00 Project sonoff Sonoff (Topic sonoff-attic, Fallback DVES_0AA47C, GroupTopic sonoffs) Version 5.14.0-2_3_0
00:00:00 WIF: Connecting to AP1 betyar0 in mode 11N as sonoff-attic-1148...
00:00:13 WIF: Connect failed with AP incorrect password
00:00:14 WIF: Connecting to AP1 betyar0 in mode 11N as sonoff-attic-1148...
00:00:17 WIF: Connected
00:00:17 DNS: Initialized
00:00:17 HTP: Web server active on sonoff-attic-1148.local with IP address 192.168.1.123
00:00:18 MQT: Attempting connection...
00:00:18 MQT: Connected
00:00:18 MQT: tele/sonoff-attic/LWT = Online (retained)
00:00:18 MQT: cmnd/sonoff-attic/POWER = 
00:00:18 MQT: tele/sonoff-attic/INFO1 = {"Module":"Generic","Version":"5.14.0","FallbackTopic":"DVES_0AA47C","GroupTopic":"sonoffs"}
00:00:18 MQT: tele/sonoff-attic/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-attic-1148","IPAddress":"192.168.1.123"}
00:00:18 MQT: tele/sonoff-attic/INFO3 = {"RestartReason":"External System"}
00:00:19 MQT: stat/sonoff-attic/RESULT = {"POWER":"OFF"}
00:00:19 MQT: stat/sonoff-attic/POWER = OFF
09:20:43 MQT: tele/sonoff-attic/STATE = {"Time":"2019-02-15T09:20:43","Uptime":"0T00:00:26","Vcc":3.058,"POWER":"OFF","Wifi":{"AP":1,"SSId":"betyar0","RSSI":70,"APMac":"00:23:CD:11:B1:9C"}}
09:20:43 MQT: tele/sonoff-attic/SENSOR = {"Time":"2019-02-15T09:20:43","Switch1":"OFF"}
09:25:43 MQT: tele/sonoff-attic/STATE = {"Time":"2019-02-15T09:25:43","Uptime":"0T00:05:26","Vcc":3.058,"POWER":"OFF","Wifi":{"AP":1,"SSId":"betyar0","RSSI":78,"APMac":"00:23:CD:11:B1:9C"}}
09:25:44 MQT: tele/sonoff-attic/SENSOR = {"Time":"2019-02-15T09:25:43","Switch1":"OFF"}
09:30:18 APP: Serial logging disabled
09:30:44 MQT: tele/sonoff-attic/STATE = {"Time":"2019-02-15T09:30:44","Uptime":"0T00:10:27","Vcc":3.058,"POWER":"OFF","Wifi":{"AP":1,"SSId":"betyar0","RSSI":76,"APMac":"00:23:CD:11:B1:9C"}}
09:30:44 MQT: tele/sonoff-attic/SENSOR = {"Time":"2019-02-15T09:30:44","Switch1":"OFF"}
09:35:44 MQT: tele/sonoff-attic/STATE = {"Time":"2019-02-15T09:35:44","Uptime":"0T00:15:27","Vcc":3.058,"POWER":"OFF","Wifi":{"AP":1,"SSId":"betyar0","RSSI":74,"APMac":"00:23:CD:11:B1:9C"}}
09:35:44 MQT: tele/sonoff-attic/SENSOR = {"Time":"2019-02-15T09:35:44","Switch1":"OFF"}

Result when GPIO is pulled HIGH

00:00:00 Project sonoff Sonoff (Topic sonoff-attic, Fallback DVES_0AA47C, GroupTopic sonoffs) Version 5.14.0-2_3_0
00:00:00 WIF: Connecting to AP1 betyar0 in mode 11N as sonoff-attic-1148...
00:00:13 WIF: Connected
00:00:13 DNS: Initialized
00:00:13 HTP: Web server active on sonoff-attic-1148.local with IP address 192.168.1.123
00:00:14 MQT: Attempting connection...
09:36:48 MQT: Connect failed to 192.168.1.104:1883, rc -2. Retry in 10 sec
09:36:59 MQT: Attempting connection...
09:36:59 MQT: Connected
09:36:59 MQT: tele/sonoff-attic/LWT = Online (retained)
09:36:59 MQT: cmnd/sonoff-attic/POWER = 
09:36:59 MQT: tele/sonoff-attic/INFO1 = {"Module":"Generic","Version":"5.14.0","FallbackTopic":"DVES_0AA47C","GroupTopic":"sonoffs"}
09:36:59 MQT: tele/sonoff-attic/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-attic-1148","IPAddress":"192.168.1.123"}
09:36:59 MQT: tele/sonoff-attic/INFO3 = {"RestartReason":"External System"}
09:37:00 MQT: stat/sonoff-attic/RESULT = {"POWER":"OFF"}
09:37:00 MQT: stat/sonoff-attic/POWER = OFF
09:37:07 MQT: tele/sonoff-attic/STATE = {"Time":"2019-02-15T09:37:07","Uptime":"0T00:00:34","Vcc":3.058,"POWER":"OFF","Wifi":{"AP":1,"SSId":"betyar0","RSSI":76,"APMac":"00:23:CD:11:B1:9C"}}
09:37:07 MQT: tele/sonoff-attic/SENSOR = {"Time":"2019-02-15T09:37:07","Switch1":"ON"}
09:42:08 MQT: tele/sonoff-attic/STATE = {"Time":"2019-02-15T09:42:08","Uptime":"0T00:05:35","Vcc":3.058,"POWER":"OFF","Wifi":{"AP":1,"SSId":"betyar0","RSSI":76,"APMac":"00:23:CD:11:B1:9C"}}
09:42:08 MQT: tele/sonoff-attic/SENSOR = {"Time":"2019-02-15T09:42:08","Switch1":"ON"}

Result when GPIO is neither HIGH or LOW

00:00:00 Project sonoff Sonoff (Topic sonoff-attic, Fallback DVES_0AA47C, GroupTopic sonoffs) Version 5.14.0-2_3_0
00:00:00 WIF: Connecting to AP1 betyar0 in mode 11N as sonoff-attic-1148...
00:00:12 RSL: stat/sonoff-attic/RESULT = {"POWER":"ON"}
00:00:12 RSL: stat/sonoff-attic/POWER = ON
00:00:12 RSL: stat/sonoff-attic/RESULT = {"POWER":"OFF"}
00:00:12 RSL: stat/sonoff-attic/POWER = OFF
00:00:13 WIF: Connect failed with AP incorrect password
00:00:14 WIF: Connecting to AP1 betyar0 in mode 11N as sonoff-attic-1148...
00:00:14 RSL: stat/sonoff-attic/RESULT = {"POWER":"ON"}
00:00:14 RSL: stat/sonoff-attic/POWER = ON
00:00:14 RSL: stat/sonoff-attic/RESULT = {"POWER":"OFF"}
00:00:14 RSL: stat/sonoff-attic/POWER = OFF
00:00:17 RSL: stat/sonoff-attic/RESULT = {"POWER":"ON"}
00:00:17 RSL: stat/sonoff-attic/POWER = ON
00:00:17 RSL: stat/sonoff-attic/RESULT = {"POWER":"OFF"}
[... 6 lines cut ...]
00:00:18 MQT: tele/sonoff-attic/LWT = Online (retained)
00:00:18 MQT: cmnd/sonoff-attic/POWER = 
00:00:18 MQT: tele/sonoff-attic/INFO1 = {"Module":"Generic","Version":"5.14.0","FallbackTopic":"DVES_0AA47C","GroupTopic":"sonoffs"}
00:00:18 MQT: tele/sonoff-attic/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-attic-1148","IPAddress":"192.168.1.123"}
[... cut ...]
09:47:20 MQT: stat/sonoff-attic/RESULT = {"POWER":"OFF"}
09:47:20 MQT: stat/sonoff-attic/POWER = OFF
09:47:30 MQT: stat/sonoff-attic/RESULT = {"POWER":"ON"}
09:47:30 MQT: stat/sonoff-attic/POWER = ON
09:47:30 MQT: stat/sonoff-attic/RESULT = {"POWER":"OFF"}
09:47:30 MQT: stat/sonoff-attic/POWER = OFF
09:47:30 MQT: stat/sonoff-attic/RESULT = {"POWER":"ON"}
09:47:30 MQT: stat/sonoff-attic/POWER = ON
09:47:31 MQT: stat/sonoff-attic/RESULT = {"POWER":"OFF"}
09:47:31 MQT: stat/sonoff-attic/POWER = OFF

Jason2866, 2019-02-15:
Do you power it with 3.3V as described? 
http://techgurka.blogspot.nl/2013/05/cheap-pyroelectric-infrared-pir-motion.html

betyar, 2019-02-15:
Yes. That's one reason I bought this type as opposed to others which are 5V. It uses the same power source as the ESP8266. You can see from the logs that the power level (according to the ESP8266 at least) is just over 3V. The setup I use is the same I use with other sensors, that is, Switch1 and a dummy Relay1. I double-checked my current setup anyway with another sensor (a photosensor to detect light and switch the dummy relay on/off accordingly) and it works no problem. The PIR appears to be working properly (it triggers) but that still doesn't mean it might not be faulty (after all it comes from China). I just thought that reading this thread others might have experienced the exact same problem. What I see is that there are some which are quite similar but still they are not the same and that's why the solutions proposed and that I tried don't seem to work (for instance I tried changing the SwitchMode and the PulseTime).

Jason2866, 2019-02-15:
So the PIR is to blame...

betyar, 2019-02-15:
I don't know. That is why I have been following this thread in case anyone has any suggestions or has experienced something similar. I have already tried the various solutions thus far mentioned but so far none have worked for me.

smadds, 2019-02-16:
Was looking at this last night - an SR501 module linking to a Wemos D1 mini. I found that the current drive out of the module was not enough to pull the Wemos pin low - it only got down to about 2.4V. I found a circuit diagram online and saw the output has a 1K resistor in series with it. Putting a 100R resistor across this improved things a lot, and the low state started being reliably sensed by the Wemos. I guess the Wemos has fairly strong pull-ups on the pins?

The output resistor is the one to the right of the output connector in the picture above, next to the GND pin and at the end of the decoder chip.

Having done all that we found that the sensitivity of the PIR module was not at all good - even with the sensitivity adjuster turned right up it would only detect body heat within a foot or so of the sensor...

betyar, 2019-02-16:
Thanks very much for your input @smadds. I'll give it a try but if the sensor can only detect body heat within a foot or so of the sensor then in the end it might not be useful for me. I planned to deploy it in the middle of a hallway which has a door on either end and a door near the middle on the side. The distance between the doors on either end and the middle is about 8 feet, so if the best the sensor can do is about a foot then it would mean having three sensors, basically one by each door. Anyway, I'll give it a try and see what happens.

gerardovf, 2019-02-16:
@betyar try this: [RCWL-0516](https://github.com/arendst/Sonoff-Tasmota/wiki/Wemos-D1-Mini-and-RCWL-0516-Microwave-Radar-Sensor)
It's working quite well for me in three sonoff basic using Tasmota rules. The only issue is that I used a 'long' (about 20 centimeters) cable to separate the radar from the sonoff interferences and also I had to implement a 4 seconds delay in node-red to get rid of short false detections.

betyar, 2019-02-16:
Thanks @gerardovf ! I'll check it out.

davepee, 2019-02-22:
Hi, I have some issue. Sonoff Basic with Tasmota FW and PIR sensor works property. Relay is turned ON when motion is detected. BUT when is connected to Wi-Fi problems begin to appear. Sonoff starts to switch ON spontaneously.

moelfus, 2020-08-02:
Hi guys,
I also have a question. I have a Sonoff 4Ch with current Tasmota FW and a AM312 attached to GPIO2. The two first channels of the 4CH are attached to two lights and the other two channels to a screen and a raspberry (MagicMirror). With the PIR I just want to send a trigger/signal to HA via MQTT without triggering the relays directly (lights shell only be switched at night, screen all day ...). Powering the channels shell be done via HA, so I just need the MQTT signal. I set up the PIR following the offical guide with:

GPIO2 -> Switch 1

And the ruleset:
`SwitchMode1 1
SwitchTopic 0
Rule1 on Switch1#state=1 do Backlog Publish stat/%topic%/PIR1 ON; RuleTimer1 30 endon on Rules#Timer=1 do Publish stat/%topic%/PIR1 OFF endon
Rule1 1`   

Using this the MQTT signal is fired when the PIR detects something, however the first channel is automatically turned on, and off again after 2 or 3 seconds. So my question is it possible to just fire the MQTT signal to HA with switching the first channel? and if so how? 
Thanks guys!

betyar, 2020-08-02:
You have to decouple the switch from the relay, in other words, create a dummy Relay5 on any unconnected GPIO and use Switch5 for GPIO2 (connected to the PIR sensor). I am assuming the relays are numbered 1-4, if not, then use any number except those used by the four existing relays.

moelfus, 2020-08-08:
> You have to decouple the switch from the relay, in other words, create a dummy Relay5 on any unconnected GPIO and use Switch5 for GPIO2 (connected to the PIR sensor). I am assuming the relays are numbered 1-4, if not, then use any number except those used by the four existing relays.

@betyar thanks, you were right. Under "Configure Module" I set GPIO2 to Switch5, changed the GPIO9 to Relay5 (and thus cannot use the button anymore, which doesn't matter) and used `switchtopic5 cmnd/motion/room/PIR1` and also set the topic as binary sensor in hassio [more details](https://community.home-assistant.io/t/how-to-get-pir-sensor-to-show-in-home-assistant-using-sonoff-with-tasmota/118109/4). So everything is working now.