arendst/Tasmota #1570: IrReceive, no code,Magic Controller

speedy16-18, 2018-01-11:
Hey,
I've uploaded Tasmota to the new Magic Home Led controller.

When I push the button I only get this message:

{"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"7C"}}

But where ist the button Code?

spooniester, 2018-01-19:
He,

some problem here....Is tasmota in general able to use the ir remote as its used with the original firmware?
Thanks

arendst, 2018-01-19:
Magic controller IR receive is not implemented.

lobradov, 2018-01-19:
Technically, IR receive is implemented: if you enable IRRecv functionality on a specific PIN, you will be able to receive IR codes from all remotes and all the keys you press are going to be sent over MQTT, just like @speedy16-18 reported. Button code is combination of protocol and data, so it's there.

What's not implemented is the **mapping** between IR keys and actions that Tasmota would take upon those keys. You have to do that in your home automation software.

On the upside, Tasmota is also able to capture non-Magic controller IR codes and you can (via your home automation) act upon them. Think of capturing any IR remote press anywhere in your apartment and using it for whatever you centrally program it for: Unused button on TV remote can be used for controlling lights.

Jason2866, 2018-01-21:
Hi lobradov,
my problem is that i always get this code:
13:17:42 MQT: tele/sonoff-87786C/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"7C"}}
13:17:42 MQT: tele/sonoff-87786C/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"7C"}}
13:17:45 MQT: tele/sonoff-87786C/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"7C"}}
13:17:46 MQT: tele/sonoff-87786C/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"7C"}}
13:17:47 MQT: tele/sonoff-87786C/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"7C"}}
13:17:52 MQT: tele/sonoff-87786C/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"7C"}}
13:17:53 MQT: tele/sonoff-87786C/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"7C"}}
13:17:55 MQT: tele/sonoff-87786C/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"7C"}}
13:17:58 MQT: tele/sonoff-87786C/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"7C"}}
13:18:00 MQT: tele/sonoff-87786C/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"7C"}}
13:18:01 MQT: tele/sonoff-87786C/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"7C"}}
13:18:03 MQT: tele/sonoff-87786C/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"7C"}}
13:18:05 MQT: tele/sonoff-87786C/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"7C"}}
I pressed different buttons on different remotes. So i think there is a fault in the decoding routine.
I do understand that i have to define the actions what to do with DIFFERENT received codes in the homautomation system ( in my case OpenHab 2).
Greets from Bavaria
J. Obermeier

lobradov, 2018-01-21:
Herr Obermeier :)

That’s definitely not how IRreceive should work.
Let me look at the newest code and try it out with few remotes I have, to see how it works.
I’m using my own old(er) code which ‘works for me’

Greetings from rainy Serbia

//_ on phone

Jason2866, 2018-01-21:
Thank you lobradov,
that was fast! I am sure you will find the bug😀

Greets from Bavaria with rain and snow

speedy16-18, 2018-01-25:
I think it is a bug, i have the same problem

speedy16-18, 2018-02-01:
Downgraded my device to Version 5.9 and now it works.

{"IrReceived":{"PROTOCOL":"NEC", "BITS":32, "DATA":"FFB04F"}}

Jason2866, 2018-02-01:
Workaround fix is here:  #1663

rwestnl, 2018-05-20:
I think it is now possible with rules but I'm not sure how to write to rule with the received data:
`{"IrReceived":{"Protocol":"NEC","Bits":32,"Data":"FFF807"}}`

https://github.com/arendst/Sonoff-Tasmota/wiki/Rules

ril3y, 2021-08-20:
Rule1 on IrReceived#Data=0xB4B8F do POWER TOGGLE endon

this worked for me.

ascillato2, 2021-08-20:
Thanks for sharing your tests!