arendst/Tasmota #1573: Changed REST status reply in latest version

Dawnes, 2018-01-11:
Not sure if intentional or not, but it seems the REST status reply has changed slightly between 5.10 and 5.11 breaking some automation for me.

http://192.168.x.y/cm?cmnd=status

New fw 5.11:
{"Status":{"Module":1,"FriendlyName":"Sonoff","Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}

Old fw 5.10:
STATUS = {"Status":{"Module":1,"FriendlyName":"Sonoff","Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}

arendst, 2018-01-13:
See releasenotes

Dawnes, 2018-01-13:
I guess that would be this one: Change output of HTTP command to valid JSON and Array only (#1363)
So change is intentional. Thanks, I'll change the detection logic on my side.