arendst/Tasmota #1575: Can't connect to WiFi with ! in the password

Fridleif, 2018-01-11:
Hi, I can't connect to a WiFi with a ! in the password. Is there a trick? With only numbers and letters are no problems.

Thanks

GrandadEvans, 2018-01-12:
I sometimes have this issue as my passwords often contain these characters. I often have to escape them, so
"Password123!"
Becomes
"Password123\!"
Sorry if this is something you are already aware of but it's worth a shot.

Fridleif, 2018-01-12:
@GrandadEvans 
Thanks for you suggestion. But it doesn't work :(

I found a solution using the WPS-button and the 5 Click option from https://github.com/arendst/Sonoff-Tasmota/wiki/Button-usage

It's more a workarround then a solution but it work for now.

closed 2018-05-26 by stale[bot]