arendst/Tasmota #1576: MagicOta question

issacg, 2018-01-12:
If I have a custom OTA setup on an external webserver, via OTA_URL, can I still use MagicOta?  What naming requirements are there?

arendst, 2018-01-12:
It should work. To select the minimal firmware version 
- I look for a dash in the OtaUrl and if it is there I replace the rest until the dot with the text "-minimal" so that an url of "../sonoff-DE.bin" becomes "../sonoff-minimal.bin"
- If there is no dash I look for the dot and append the text "-minimal" so that an url of "../sonoff.bin" becomes "../sonoff-minimal.bin"

I do not check slashes so forward or backward slashes should work the same.

issacg, 2018-01-15:
marking closed - my firmware isn't big enough to test this yet.  I'll reopen if needed.

I still note that my minimal comes out to over 350k (with MQTT_TLS, and everything else I felt safe disabling, disabled)

closed 2018-01-15 by issacg