arendst/Tasmota #1577: Sonoff T1 UK - Serial Connection

juan11perez, 2018-01-12:
Good day. The serial connection picture in here (https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-T1-UK-and-T1-EU) is incorrect.

This one is correct (https://user-images.githubusercontent.com/25724469/33124356-89e19e78-cf74-11e7-8e8b-c0a946d58884.jpg)

juan11perez, 2018-01-12:
Don’t know but I flashed 10 T1s today and they were all like the picture I posted.

hnee4711, 2018-01-12:
Both pictures are essentially the same. The first one shows Tx needs to connect to Rx, so it is more complete. If you ignore look at the connection only at the soldering points, then  both pictures are the same. This should be closed

geekypenguin, 2018-01-13:
Agreed, this can be closed. above information will be included in the wiki update i will be making later today