arendst/Tasmota #1578: Sonoff Dual - Echo Plus with Built-in Hub

zibous, 2018-01-12:
I can not manage Sonoff Dual with Alex (Echo Plus with Built-in Hub). It works with the WeMo emulation, but then I can only control one relay. Echo does not recognize a device with Hue Bridge (multi device).

[image]

Is there a workaround or a tip on how I could control the shutters with Sonoff Dual via Amazon Echo?

cyclemat, 2018-01-15:
Ich würde es über open hab steuern problem ist du müsstest so ein deinem Relais einstellen können das immer nur eins an ist das kann die Tasmota Firmware Bisher nicht da du ja scheinbar über relais 1 Rauf und Relais 2 Runter Fährst. Ich würde das ganze einfach als 2 Devices in Openhab anlegen und per MQTT Steuern Openhab hat auch Hue emulation eingebaut von daher super sache und du kannst auch Setze auf 50% usw nutzen.
Zweite idee die ich hätte alles über ein Relais Steuern also über NC und NO so kannst du aber nur komplett rauf oder komplett runter fahren.

zibous, 2018-01-16:
Via a web interface, this is easy with the home assistant application. There I can use "UP, STOP, CLOSE". Interesting would be a controller with Amazon Echo / Dot devices.

[image]

I am looking for a general skill with which I can pass any voice commands to the MQTT Brocker and then control the Sonoff devices with NODE-RED. But so far have found no approach / solution.

[image]

thomasnordquist, 2018-01-16:
I use https://www.npmjs.com/package/node-red-contrib-alexa-local, but it is very limited in functionality.

It basically emulates a philips hue bridge.
Commands are something like:
 - Turn {DeviceName} on
 - Change the temeprature of {DeviceName} to 21 ~degrees~ 
 *(no unit allowed here, actually changing the brightness)*

It does not yet support change the light color, I'm trying to change that with a pull request within in the next weeks.

To control things like AVRs, thermostats etc you'll have to use a "cloud" solution, the solution for node-red is hosted by some random dude.
https://www.npmjs.com/package/node-red-contrib-alexa-home-skill
Sadly the server side is not open source as far as I can tell, but it shouldn't be to hard to write your own backend if you are familiar with javascript, amazon has some very nice boilerplate code.

Worth mentioning:
https://flows.nodered.org/node/node-red-contrib-alexa

zibous, 2018-01-16:
@thomasnordquist, thanks for this information.

**node-red-contrib-alexa-local** can not do much more than the tasmota WeMo or philips hue bridge emulation and with this i can not control shutters because they have 3 states (ON, STOP, OFF).

- Alexa open kitchen shutter
- Alexa stop kitchen shutter
- Alexa close kitchen shutter
- Alexa close kitchen shutter 50%
.....

Yes, **node-red-contrib-alexa** is promising, but I have to create a skill and I have not done that yet (just try to learn how i can build a skill).

Nice would be a "MQTT generic Skill" that passes the voice commands to a MQTT Brocker, then they could be queried and processed with Node-RED conditions.

Am still testing and looking for ...

closed 2018-05-30 by stale[bot]