arendst/Tasmota #1581: inline light switch sonoff basic or sonoff th10

andyd999, 2018-01-12:
Hi Guys
I`m new to sonoff and cant seem to find a decent user forum.  What i basically want to do is add an inline sonoff in parallel to a standard light switch to give me the option of remotely turning lights on/off or using google home voice with it.  How do I achieve it? could i somehow modify the sensor lead for the sonoff th10 to do a chaange of state and get the sonoff to turn the lights on/off

hope the above kinda makes sense

geekypenguin, 2018-01-13:
Sonoff T1 is exactly what you need

andyd999, 2018-01-13:
The issue is I want to use the existing stylish light switches that match the decor. Unfortunately i find the tough sonoff light switches to look cheap. So I want to use the original switch but use that as a kind of relay?

geekypenguin, 2018-01-13:
I see. 
You have a similar issue to what I had and settled on the T1s. The issue is that, regardless of how they are wired, the existing wall switch will not play nicely with the sonoff. If wired in parallel then the physical switch being on, prevents the sonoff turning off. Similarly with series wiring, they would both have to be on to function.

the only other way I can see of wiring it would be to modify a sonoff basic so that your wall switch is now connected to the GPIO-0 button with the lighting circuit permanently powering the sonoff (ie the light switch itself is no longer part of the mains circuit).  You would then need to change the button mode to FOLLOW in user_config.h

andyd999, 2018-01-13:
What you put at the bottom is perfectly what I saw briefly somewhere. How would you modify the sonoff and would you need to change the firmware? Any way of using the sensor input on the th10 model to use gpio

geekypenguin, 2018-01-13:
Sorry I assumed you were asking based on changing the firmware already. The above I mentioned is based on flashing the sonoff to use Tasmota. I dont know how you would do this with the stock firmware. It would then be fairly trivial to configure the connection for the sensor to be the switch instead as this already goes to one of the GPIO. You would just need to change the input type to switch in FOLLOW mode as before. I believe this can be done via the web interface but I dont have one to test myself

andyd999, 2018-01-13:
So I could put tasmota on a sonoff th10 then wire the light switch into the
2.5mm jack plug and configure the software ?
Also adding tasmota software will make the device compatible with Samsung
smartthings hub?

geekypenguin, 2018-01-13:
Yes that is correct I believe.
I don't know if Tasmota makes it compatible with Samsung smartthings. Tasmota supports Hue and WeMo emulation (so works with google home and amazon echo etc) and MQTT

andyd999, 2018-01-13:
Thanks for your help. I'll get a new sonoff th10 and then flash with tasmota and have a play. The other issue I've just noticed Is our wall socket boxes aren't deep enough for tge t1 switches so the wall would need drilling out deeper and a replacement back box fitting. Fingers crossed the th10 will work

geekypenguin, 2018-01-13:
Not sure if the Th10 will fit either then but good luck all the same!

andyd999, 2018-01-13:
No it definitely won't fit it will have to go into the ceiling void

mscreations, 2018-01-14:
I did this exact thing with a sonoff basic. I soldered a couple wires to the ground and Gpio14 through hole pins and attached then to my toggle switch. Then in the configuration page, you set gpio14 to switch and it'll work good. Plus it was small enough to fit behind the switches in my box.

Frogmore42, 2018-01-14:
While I believe Tasmota and Sonoffs are great, for what you said you wanted I would use something like this: https://www.amazon.com/Vision-Z-Wave-Micro-Switch-relay/dp/B00R883YKU/ref=sr_1_3?ie=UTF8&qid=1515956442&sr=8-3&keywords=in+wall+zwave

This is an in-wall Z Wave switch that is designed to do exactly what you want.  You will need a z wave controller (Samsung SmartThings is one).  Most electrical boxes are space constrained and not big enough to hold a Sonoff.

andyd999, 2018-01-14:
I'm from the uk and our light switch boxes are like 70mm square and 35mm deep so sonoff definitely wouldn't fit if mount it under the floor above the switch cables. But your idea s even better I'll see if there is a uk supplier

closed 2018-05-28 by stale[bot]