arendst/Tasmota #1585: Sonoff basic with pull up resistor.

echo4019, 2018-01-13:
Hi, 

I connected a light and an external switch (toggle) to a sonoff basic using the gpio14.
However the light randomly goes on and off, it seems the long wire length to the switch is causing interference. 

I have added a pull-up resistor connected between the GPIO14 pin and 3.3v. This has solved the problem, but when the switch is in the closed position the resistor rapidly heats up. The resistor i have used is a 10 ohm 2W Metal Film Resistor 10R 2 Watt - http://www.ebay.co.uk/itm/5-X-10-ohm-2W-Metal-Film-Resistor-10R-2-Watt/180709367238?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649. 

Is this the wrong type of resistor? 

Any help is greatly appreciated.

thomasnordquist, 2018-01-13:
You should probably use a resistor something between 1-10k Ohm.

The reason why it gets warm is that you burn a lot of energy there.
I = U / R
I = 3.3V / 10 Ohm = 0.33A
So you got a current of approximatly 0.33A, burning somthing around 1W (0.33A * 3.3Volt).
It's perfectly okay for the resistor, and it's normal that it becomes very very warm.

If GPIO14 is at some time not in a high impedance state (input), you might exceed the maximum rated drive current of the microcontroller, which is rated at 12mA. To prevent that you should get a resistor >= 275Ohm

Frogmore42, 2018-01-14:
To put it more simply, when you want a pull-up resistor on a digital pin for a manually controlled switch, start with a 10K ohm resistor.  You can use a normal 1/8 or 1/4W resistor.  The exact value is not that important so even a 10% one would be fine.

I2C and other communication protocols that use a pull-up typically use other (smaller) values because they need to switch values faster.

If you want to save power use an even higher resistance value, but that might have issues with your very long lines.

MPJK77, 2018-03-08:
Hi, 
I am having the same issue.
_I connected a light and an external switch (toggle) to a sonoff basic using the gpio14.
_The light randomly goes on and off, it seems the long wire length to the switch is causing interference.__
How and where do I add a pull-up resistor? 

Thanks in advance!

Frogmore42, 2018-03-08:
Normally, the pull up resistor is connected near the input pin.  So, solder a 10K resistor between GPIO 14 and 3V3.  Sparkfun has a tutorial on pull-up resistors that might be worth reading.

MPJK77, 2018-03-08:
Thanks for the answer, I will check it out on sparkfun and solder the resistor into the sonoff!

deviant-studio, 2018-10-21:
i spent a whole day until realized that 10kOhm resistor was the reason of my sonoff basic randomly restarts and connections losts. i changed it to 4.7kOhms and stability has been restored. hope this helps somebody.