arendst/Tasmota #1586: Console ignores topic

matsekberg, 2018-01-13:
When typing in the console, I don't have to enter the correct topic for a command:

16:07:58 CMD: status/**aaaaaaaaa**/POWER toggle
16:07:59 MQT: status/th-0681/RESULT = {"POWER":"on"}
16:07:59 MQT: status/th-0681/POWER = on
16:08:07 CMD: status/**wedwedwedwedwedewdewd**/POWER toggle
16:08:07 MQT: status/th-0681/RESULT = {"POWER":"off"}
16:08:07 MQT: status/th-0681/POWER = off

Is that correct behavior?

arendst, 2018-01-13:
Yep. You're already working on the device so there is no need to enter a topic.

matsekberg, 2018-01-13:
Ok.

I have another strange behavior, this is messages as logged by mosquitto:

sensor/th-0681/STATE {"Time":"2018-01-13T16:11:50","Uptime":0,"Vcc":3.054,"POWER":"off","Wifi":{"AP":1,"SSId":"SSID2_EXT","RSSI":88,"APMac":"B0:7F:B9:3E:E1:63"}}
sensor/th-0681/SENSOR {"Time":"2018-01-13T16:11:50","DS18B20":{"Temperature":17.5},"TempUnit":"C"}

Then I send "cmnd/th-0681/POWER toggle" with mosquitto_pub and also tried node-red. This is the log in mosquitto, and the relay does NOT toggle:

cmnd/th-0681/POWER toggle
cmnd/th-0681/POWER toggle

When I hit the button on the Sonoff-TH I get:

status/th-0681/RESULT {"POWER":"on"}
status/th-0681/POWER on
status/th-0681/RESULT {"POWER":"off"}
status/th-0681/POWER off

So why cant I send "cmnd/th-0681/POWER toggle" and get the relay to toggle??

arendst, 2018-01-13:
For mosquitto you are using -t for topic (../POWER) and -m for payload (toggle)?

matsekberg, 2018-01-13:
` mosquitto_pub -h 10.0.1.50 -t "cmnd/th-0681/POWER" -m "toggle"`

matsekberg, 2018-01-13:
When using node-red this gets in the mosquitto log:
`cmnd/th-0681/POWER toggle`

arendst, 2018-01-13:
Somewhere you changed the default ``stat`` name to ``status``. I guess you did the same with the default ``cmnd`` name ...

My mosquitto works as expected:
```
mosquitto_pub -h domus1 -t "cmnd/th1/POWER" -m "toggle"
```
Check your prefixes with commands ``prefix1``, ``prefix2`` and prefix3``

matsekberg, 2018-01-13:
Your a genious !!
I certainly did, and since the settings are preserved through upgrades....
Thanx!

```
16:27:26 CMD: prefix1
16:27:26 MQT: status/th-0681/RESULT = {"Prefix1":"action"}
16:27:29 CMD: prefix2
16:27:29 MQT: status/th-0681/RESULT = {"Prefix2":"status"}
16:27:30 CMD: prefix3
16:27:30 MQT: status/th-0681/RESULT = {"Prefix3":"sensor"}
```

closed 2018-01-13 by matsekberg