arendst/Tasmota #1587: Security issue ? wireless information not protected

matsube11, 2018-01-13:
Sonoff-Tasmota: 5.11.1 20180107
Sonoff wifi

Hi,

I realized today while I was rebooting my wireless access point that my Sonoff device once disconnected was showing the wifi parameters page without requiring any password. Even though I setup a web admin password, password is only requested if changes are to be made on the page.

So it's trivial to get the wifi SSID and password information from someone using a sonoff tasmota, just disconnect it from wifi, connect to it and that's it.

Am I missing something here ???

Thanks

davidelang, 2018-01-13:
change your wifi mode so that it doesn't present an AP if it fails to connect to 
one.

matsube11, 2018-01-13:
Thanks, changed successfully using WifiConfig 4

I'm new to Tasmota, this is amazing piece of software but this default behavior is a critical security breach imho.
Wonder how many people have their wifi crendentials open for anyone to access due to that :/

ascillato, 2018-01-14:
Hi,

I test this issue and this allows you to change the Wifi configuration so you can take yours neighbour's sonoff and connect it to your network. In that initial page you only can see the SSID but not the password.

After that, you can not control it. The Sonoff ask you for the user and password that you previously configure.

So, this is just an anoying issue that can take your sonoff from your network.

I consider that a possible solution is to automatically change the WifiConfig variable to WIFI RECONNECT after the sonoff connect to a network. Then you should be able to configure the WifiConfig on the Webserver Menus.

I can take a look on doing this if this solution is considered ok.

davidelang, 2018-01-14:
and what if someone wants their device set this way? how to they avoid your "I 
know what's best for you" feature?

ascillato, 2018-01-14:
The logic I follow was:

1- On the first time you power up the sonoff, it shows as AP so you can set up your WIFI with your phone
2- Sonoff connects to your Wifi
3- Now you can set up the features of your sonoff over your wifi.

From step 2 to 3, Sonoff can change the
    WIFI_CONFIG_TOOL: WIFI_MANAGER
to
    WIFI_CONFIG_TOOL: WIFI_RETRY

Because now your Sonoff is attached to your Network. Each time it get disconnected from your router, will reconnect to it. But, if you want that each time the sonoff get disconnected from your wifi the sonoff shows itself as AP,, you should be able to configure that from the EXTRA CONFIG menu of the webserver.

What do you think?

Another thing that I tested that is working very well on the actual version (5.11.1 20180107) is that if you push reset 3 times, the sonoff uses the factory settings. So if you flash the sonoff with WIFI_CONFIG_TOOL: WIFI_MANAGER, you can now take control again without a router.

davidelang, 2018-01-14:
how do you get to the device if the network it's trying to connect to no longer 
exists? you can't get to the webserver to set it.

If you want to do this, create a new WIFI_MANAGER_ONCE mode and leave the 
existing WIFI_MANAGER alone.

(or you could create WIFI_MANAGER_I_REALLY_MEAN_IT mode that works the way 
WIFI_MANAGER does today)

ascillato, 2018-01-14:
Yes, _WIFI_MANAGER_ONCE_ is what I meant.

I found that the variable **Settings.sta_config** stores the value of _WIFI_CONFIG_TOOL_.

So this mode (_WIFI_MANAGER_ONCE_) just only need to change **Settings.sta_config = WIFI_MANAGER** to  **Settings.sta_config = WIFI_RETRY**

But also you should be able to change this from the Config Extra Menu

ascillato, 2018-01-14:
After testing some code to solve this issue, and also with the feedback from @davidelang, I think that the best solution to this is that: If the **web admin password** is set, then, when you try to connect to the Sonoff AP, it asks you for that password.

What do you think? I will try this.

matsube11, 2018-01-14:
I would have expected what @ascillato suggests, meaning that access to Sonoff AP would be password protected.

> I test this issue and this allows you to change the Wifi configuration so you can take yours neighbour's sonoff and connect it to your network. In that initial page you only can see the SSID but not the password.

Just use ShowPassword for Chome for example and you'll see that have instant access to the wifi password as well :/

ascillato, 2018-01-14:
Changes proposed for: **_webserver.ino_** so as to solve the security issue #1587 from @matsube11 (Security issue ? wireless information not protected)

 -Stop sending password information to client
 -Credenditals ask if are set when Sonoff is on AP mode.

**1- Never send password information to client.** Instead send 10 asterisks.

Line 207
 original:
```
  "<br/><b>" D_AP1_PASSWORD "</b><br/><input id='p1' name='p1' type='password' placeholder='" STA_PASS1 "' value='{p1'><br/>"
```
 proposed:
```
  "<br/><b>" D_AP1_PASSWORD "</b><br/><input id='p1' name='p1' type='password' placeholder='" D_AP1_PASSWORD "' value='{p1'><br/>"
```

Line 209
 original:
```
  "<br/><b>" D_AP2_PASSWORD "</b><br/><input id='p2' name='p2' type='password' placeholder='" STA_PASS2 "' value='{p2'><br/>"
```
 proposed:
```
  "<br/><b>" D_AP2_PASSWORD "</b><br/><input id='p2' name='p2' type='password' placeholder='" D_AP2_PASSWORD "' value='{p2'><br/>"
```

Line 239
 original:
```
  "<br/><b>" D_WEB_ADMIN_PASSWORD "</b><br/><input id='p1' name='p1' type='password' placeholder='" WEB_PASSWORD "' value='{p1'><br/>"
```
 proposed:
```
  "<br/><b>" D_WEB_ADMIN_PASSWORD "</b><br/><input id='p1' name='p1' type='password' placeholder='" D_WEB_ADMIN_PASSWORD "' value='{p1'><br/>"
```

Line 790
 original:
```
  page.replace(F("{p1"), Settings.sta_pwd[0]);
```
 proposed:
```
  page.replace(F("{p1"), "**********");
```

Line 792
 original:
```
  page.replace(F("{p2"), Settings.sta_pwd[1]);
```
 proposed:
```
  page.replace(F("{p2"), "**********");
```

Line 886
 original:
```
  page.replace(F("{p1"), Settings.web_password);
```
 proposed:
```
  page.replace(F("{p1"), "**********");
```

**2- The original code that save the passwords, have a comparison to see if you enter nothing and, if that the case, it will put the default password. I think that if you put nothing, the password should be nothing as I learnt from @davidelang to _avoid the "I know what's best for you" features_ hehe :)
     Also, the following code has a comparison to see if the password has asterisks so as not to save anything**

Line 967 to 972
 original:
```
    strlcpy(Settings.sta_ssid[0], (!strlen(WebServer->arg("s1").c_str())) ? STA_SSID1 : WebServer->arg("s1").c_str(), sizeof(Settings.sta_ssid[0]));
    strlcpy(Settings.sta_pwd[0], (!strlen(WebServer->arg("p1").c_str())) ? STA_PASS1 : WebServer->arg("p1").c_str(), sizeof(Settings.sta_pwd[0]));
    strlcpy(Settings.sta_ssid[1], (!strlen(WebServer->arg("s2").c_str())) ? STA_SSID2 : WebServer->arg("s2").c_str(), sizeof(Settings.sta_ssid[1]));
    strlcpy(Settings.sta_pwd[1], (!strlen(WebServer->arg("p2").c_str())) ? STA_PASS2 : WebServer->arg("p2").c_str(), sizeof(Settings.sta_pwd[1]));
```
 proposed:
```
    strlcpy(Settings.sta_ssid[0], (!strlen(WebServer->arg("s1").c_str())) ? "" : WebServer->arg("s1").c_str(), sizeof(Settings.sta_ssid[0]));    
    strlcpy(Settings.sta_pwd[0], (!strlen(WebServer->arg("p1").c_str())) ? "" : (strchr(WebServer->arg("p1").c_str(),'*')) ? Settings.sta_pwd[0] : WebServer->arg("p1").c_str(), sizeof(Settings.sta_pwd[0]));
    strlcpy(Settings.sta_ssid[1], (!strlen(WebServer->arg("s2").c_str())) ? "" : WebServer->arg("s2").c_str(), sizeof(Settings.sta_ssid[1]));
    strlcpy(Settings.sta_pwd[1], (!strlen(WebServer->arg("p2").c_str())) ? "" : (strchr(WebServer->arg("p2").c_str(),'*')) ? Settings.sta_pwd[1] : WebServer->arg("p2").c_str(), sizeof(Settings.sta_pwd[1]));
```

Line 1018 
 original:
```
    strlcpy(Settings.web_password, (!strlen(WebServer->arg("p1").c_str())) ? WEB_PASSWORD : (!strcmp(WebServer->arg("p1").c_str(),"0")) ? "" : WebServer->arg("p1").c_str(), sizeof(Settings.web_password));
```
 proposed:
```
strlcpy(Settings.web_password, (!strlen(WebServer->arg("p1").c_str())) ? "" : (strchr(WebServer->arg("p1").c_str(),'*')) ? Settings.web_password : WebServer->arg("p1").c_str(), sizeof(Settings.web_password));
```

**3- If there is a password set and you are not authenticated, then ask for credentials. (if there is a password set and you have WIFI_CONFIG_TOOL = WIFI_MANAGER on _"user_config.h line 62"_, when Sonoff disconnects from wifi network it changes to AP. You can connect to that AP and it ask for user and password to continue / otherwise, if you do not have password set, it just shows the wifi config page)**

Line 414
 original:
```
  if((HTTP_ADMIN == webserver_state) && (Settings.web_password[0] != 0) && !WebServer->authenticate(WEB_USERNAME, Settings.web_password)) {
```
 proposed:
```
  if((Settings.web_password[0] != 0) && !WebServer->authenticate(WEB_USERNAME, Settings.web_password)) {
```

**obs:** _When the web password is NOT SET, connecting with a phone to the AP will show automatically the captive webpage <-- cool :)
     When the web password is SET, connecting with a phone to the AP will NOT show anything. You have to manually open web browser and type 192.168.4.1 and after that it asks you for User and Pass to show the config webpage.
     If you use a PC you have to open the browser in both cases and type 192.168.4.1 to show the config webpage._

Tested Ok.

What do you think of all this approaches?

What about having an initial page asking for credentials instead of the pop up asking for credentials?

ascillato, 2018-01-14:
Another thing,

what about the **WIFI_MANAGER_ONCE** mode? (**WIFI_MANAGER mode** then after connection -> **WIFI_RETRY mode**)

I think could be handy. The first time you turn Sonoff on, it will ask you for the Wifi Settings to connect. Then, after connected to your wifi, if it disconnects will try to connect. If the case is that you change your wifi settings, you can keep pushed at any time a reset button on the sonoff (i.e. GPIO14) and it will reset to default settings, that would be WIFI_MANAGER_ONCE mode.

I saw that other devices like Chromecast when they lost connection they pass to AP mode, while others like Bose Soundtouch Speakers do something like WIFI_MANAGER_ONCE and also have buttons to change to AP as needed for reconfigure.

Just 2 features should be needed:

_-Add WIFI_MANAGER_ONCE mode
-Add an option to choose any button as also reset button on Module Configuration._

So, what do you think about this?

ascillato, 2018-01-15:
Just tested an initial page asking for credentials to be used by the captive portal, and works better than the previous approach I post.

The webserver.ino file for the previous pull request was updated with this change.

Hope to be useful.

ascillato, 2018-01-15:
Screenshots of this changes:

If you set the web admin password, when you connect to the AP you will see:

[image]

so,

[image]

then,

[image]

so you can save your SSID and PASS and will connect to it.

stefanbode, 2018-01-16:
I don't know when I checked the AP mode past time, but there you get to the normal webpage and could configure everything. Including having the webpassword request. The current behavior is for sure a leak and a second proof that for IoT devices you should have a separate network :-).

@ascillato I really like all your changes. Additionally, I would vote also for the AP mode that we use the normal UI and not the stripped down one currently available

matsube11, 2018-01-16:
Wow, impressed to see a nice fix coming that fast

Thanks @ascillato and all

Ivoz, 2018-02-05:
Couldn't this also be solved by having the 'web admin' password actually be a password for the webserver to enable HTTP Basic Authentication?

ascillato, 2018-02-05:
The issue related to **Login** and **sending the wifi password to the web page** is already solved on _Tasmota v5.11.1i_

The only issue solved on PR#1727 that is not included on current Tasmota version is that Tasmota is sending MQTT password to the web page.

richrd, 2018-02-18:
Why can't we add a WIFI password to the actual AP itself? Seems like the way to go IMHO. Why let anyone and everyone even get in to the AP?

Ivoz, 2018-02-18:
@richrd maybe some people want to allow guests on the AP, but not to be able to read server daemon logins just by visiting the web server of an IP that's also on the AP.

richrd, 2018-02-19:
@Ivoz Sounds reasonable. Could we add an option for a AP password for those that want it? I could happily write a PR for that.

closed 2018-06-21 by ascillato2