arendst/Tasmota #1588: Sonoff Basic switches on but not off through Home Assistant

dinger69, 2018-01-13:
I cannot get the Home Assistant to turn the switch off after I turn it on. After pressing the toggle switch in the Home Assistant UI for the switch, it turns it on and then the toggle switch in the UI resets to the off position yet the switch remains on.

switch:
 platform: mqtt
 name: "Bedroom Switch"
 state_topic: "stat/sonoff1/power"
 command_topic: "cmnd/sonoff1/power"
 payload_on: "ON"
 payload_off: "OFF"
 optimistic: false
 qos: 1
 retain: true

terba, 2018-01-13:
Change power to POWER

dinger69, 2018-01-13:
That fixed it. Thanks terba!!!

closed 2018-01-14 by arendst

HK-Moebius, 2018-12-03:
I´ve the same problem, but only with two sonoff basic´s

mqtt:
  username: homeassistant
  password: !secret mqtt_password
  
switch:
  - platform: mqtt
    name: "Wohnzimmerlampe"
    command_topic: "cmnd/Wohnzimmerlampe/power"
    state_topic: "stat/Wohnzimmerlampe/POWER"
    qos: 1
    payload_on: "ON"
    payload_off: "OFF"
    retain: true

  - platform: mqtt
    name: "Vitrine"
    command_topic: "cmnd/Vitrine/power"
    state_topic: "stat/Vitrine/POWER"
    qos: 1
    payload_on: "ON"
    payload_off: "OFF"
    retain: true

  - platform: mqtt
    name: "Workshop"
    command_topic: "cmnd/Workshop/power"
    state_topic: "stat/Workshop/POWER"
    qos: 1
    payload_on: "ON"
    payload_off: "OFF"
    retain: true

  - platform: mqtt
    name: "Ankleide"
    command_topic: "cmnd/Ankleide/power"
    state_topic: "stat/Ankleide/POWER"
    qos: 1
    payload_on: "ON"
    payload_off: "OFF"
    retain: true

  - platform: mqtt
    name: "Bett"
    command_topic: "cmnd/Bett/power"
    state_topic: "stat/Bett/POWER"
    qos: 1
    payload_on: "ON"
    payload_off: "OFF"
    retain: true

the last two sonoff basic´s ("Ankleide" and "Bett") can only switched on trough homeassistant. The switch moves after clicking back to off but the lamp keeps on. All Sonoff´s are on the same firmware 6.3.0 

[image]

I´ve no idea what´s going wrong the other sonoff´s are working fine expect "Ankleide" and "Bett"
I checked the settings in the tasmota webinterface several times and it´s all the same.

terba, 2018-12-03:
I would try what I wrote in the first comment. So change **all** the topic names to uppercase `power.`

HK-Moebius, 2018-12-03:
Already done, but no change. I will try mosquito and not be build in broker. Hope this will help.

ascillato, 2018-12-03:
Hi, you are using retain true. That makes lots of issues. Please, check the Tasmota wiki at home assistant chapter for the correct way of setting your devices. Thanks