arendst/Tasmota #1593: Not able to manage Magic Home 2.0 LED Strip Controller

joerg-janes, 2018-01-13:
Good Evening!

I have flashed a Magic Home Controller Board version 2.2. The Tasmota system is working! Great - Thanks for that!

Here my template

   { "MagicHome 2.0",   // Magic Home 
      0,                // GPIO00
      0,                // GPIO01
      GPIO_LED1_INV,    // GPIO02 Blue onboard LED
      0,                // GPIO03
      GPIO_USER,        // GPIO04 IR receiver (optional)
      GPIO_PWM1,        // GPIO05 RGB LED Red
      0,                // GPIO06
      0,                // GPIO07
      0,                // GPIO08
      0,                // GPIO09
      0,                // GPIO10
      0,                // GPIO11
      GPIO_PWM2,        // GPIO12 RGB LED Green
      GPIO_PWM3,        // GPIO13 RGB LED Blue
      0,                // GPIO14
      GPIO_USER,        // GPIO15 RGBW LED White (optional - set to PWM4 for Cold White or Warm White)
      0,                // GPIO16
      0                 // ADC0
   }

The GPIO numberes are from espurna, with which the controller is working fine ...

With Sonoff-Tasmota 5.11.1, I have some problemes:

There is no button to switch on or off the LED strip? As far as I read in the forum, this should be right?

But how do I control the LED strip?

As far as I read, I will have to use the MQTT commands ... but there are my problems ... please have a look to my console:

22:52:04 CMD: /homenet/led02/cmnd/Pwm1 = 1023
22:52:04 MQT: /homenet/led02/stat/RESULT = {"PWM":{"PWM1":0,"PWM2":0,"PWM3":0}}
22:53:31 CMD: /homenet/led02/cmnd/Pwm1 = 255
22:53:31 MQT: /homenet/led02/stat/RESULT = {"PWM":{"PWM1":0,"PWM2":0,"PWM3":0}}
22:54:13 CMD: /homenet/led02/cmnd/PwmRange
22:54:13 MQT: /homenet/led02/stat/RESULT = {"PWMRange":1023}
22:58:36 CMD: /homenet/led02/cmnd/SetOption15
22:58:36 MQT: /homenet/led02/stat/RESULT = {"SetOption15":"OFF"}
22:58:49 CMD: /homenet/led02/cmnd/SetOption15 = ON
22:58:49 MQT: /homenet/led02/stat/RESULT = {"SetOption15":"OFF"}

The pwm does not change and I cannot change to the color commands with SetOption15

How is the right way to manage the Magic Home Controller?

Thanks in advance for your help!
Joerg

arendst, 2018-01-13:
almost there. Try setoption15 on without the =

joerg-janes, 2018-01-14:
Thanks for your fast answer - now my MagicHome Controller 2.2 is manageable!

closed 2018-01-14 by arendst